bigtreetech/BIGTREETECH-SKR-V1.3 #87: Can't control part cooling fan

bnemmers, 2019-09-28:
I'm using the "CNC Fan" pins for the Hot end fan, and using the "12/24v Out" for the part cooling fan. But cannot turn off the part cooling fan, nor control the speed.

Thanks in advance for any help, using Marlin-bugfix-2.0.x

tvixen, 2019-09-30:
Any good reason why you do that ? (2 extruders Etc.)
Normally specs says Hotend fan 12/24V and Partfan E1 (with one extruder) as E1 is PWM controlled.

But upload you config files, and sets have a look at it.

bnemmers, 2019-09-30:
I turn on the extruder fan only when the Hotend is above 50C. This way when the printer just idle, the fan is not running. So I'm trying to find a way to control both fans via g-code. I use ABS most of the time so I don't need the parts cooling fan on. So every time I run ABS, I need to unplug the cooling fan. I beginning to find out that this method my not be and option, unless I find a spare pin to use and control it via a relay.

GForce2010, 2019-10-01:
The 12/24V out is a permanent live and cant be controlled.

You should plug the part cooling fan into the FAN port next to HE1.

You can then wire the hotend fan into HE1 and use that to turn the hotend fan off below 50C bu changing this line in the "configuration_adv.h" file

#define E0_AUTO_FAN_PIN P2_04

Marlin will then use HE1 output for the Hotend fan and will turn it on above 50C and off below that.

Tomthi, 2019-10-24:
Is there another way to control the part cooling fan? I have dual extruders and need the HE1 for the second hot end.

GForce2010, 2019-10-24:
You might be able to use a spare I/O pin like an unused min or max endstop pin with an external mosfet. Dont know if this is possible but thats where i would start my research anyway.

Tomthi, 2019-10-25:
Thanks, that is what I was thinking. Aside form the endstop pins, are there other available pins?

ipatch, 2019-11-12:
@bnemmers 

I had the exact same issue, but I ended up getting it sorted out.  I'm controlling the part cooling fan using pin 2.3 and can turn the fan on and off through the tft35 v2.0.

It does take some tweaking via marlin.  I put a little video of my troubleshooting experience on youtube [here](https://www.youtube.com/watch?v=oD1DJbHIX3o) and my updated configuration that ended up solving the issue [here](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/103#issuecomment-549160742)

Demxsr, 2020-05-12:
I was having this issue. Part cooling fan wouldn't turn on while print was going, but I could turn it on via the screen and control it from there. After spending all night going through every possibility and bug in the firmware, I finally found the issue was with Cura. Apparently there is a bug if you're using a new version of Cura and an old version of Marlin.  Under machine settings for your printer, you need to go to the Extruder tab and set "Cooling Fan Number" from 1 to 0. This completely fixed my issue.

Esla2, 2021-02-01:
Hi All,
I came here to share a trick. 
I’m running BTT SKR V1.4 Turbo with 2 extruders and using Fan0 for cooling my 2 Hotends.
- My 2 Fans (24V) are in parallel connected to the board on FAN0 (2.3) 
- HE0 (2.7) and HE1 ( 2.4) connected to the Hotends.
In this post there is another solution : https://github.com/MarlinFirmware/Marlin/issues/17455
But I found my second Hotend( HE1) don’t heated quickly as HE0.
My trick is to “deactivate” P2_04 like a fan_pin !
- In pins_BTT_SKR_common.h : I comment FAN_PIN :
//#ifndef FAN_PIN
//  #define FAN_PIN                          P2_04
//#endif

- In Configuration_adv.h : 
#define E0_AUTO_FAN_PIN  P2_03
#define E1_AUTO_FAN_PIN  P2_03

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 230 //90%  // 255 == full speed 

Advantage: 
- The Fans spin at 50° and stop below 50° automatically.
- Fans spin if either one of Hotends is turn on, or the two.
Disadvantage: 
- I can’t control my fans through my TouchScreen.

tvixen, 2021-02-02:
>   //#ifndef FAN_PIN
>   //  #define FAN_PIN                          P2_04
>   //#endif
> * In Configuration_adv.h :
>   #define E0_AUTO_FAN_PIN  P2_03
>   #define E1_AUTO_FAN_PIN  P2_03
> 
> #define EXTRUDER_AUTO_FAN_TEMPERATURE 50
> #define EXTRUDER_AUTO_FAN_SPEED 230 //90% // 255 == full speed
> 

Great idea ... it could become handy. Thanks