bigtreetech/BIGTREETECH-SKR-V1.3 #88: SD card and 12864 lcd display connection.

shtv-electro, 2019-10-01:
Hello! Faced a problem how it is possible to connect sd reader (SPI) and lcd 12864 specified on a photo. To use the module don't plan because the dimensions do not match. The documentation did not find Pinout conclusions EXP1.
[image]
[image]

CsepregiArtur, 2019-10-03:
hello

do you have these parts already? 

if not search something more compatible, otherwise search google, but also my question is from where will you control your printer? I don't see any buttons....

karabas2011, 2019-10-15:
It does not differ from any embedded cardreaders. Connect to ext2
There is no cddetect output pin so you need to solder it yourself.