bigtreetech/BIGTREETECH-SKR-V1.3 #90: Skr 1.3 WiFi

grego1981, 2019-10-02:
Hi all,

I really enjoy my SKR1.3 it works flawlessly! Is it possible to connect it to wifi via esp module? I hear some people have done it but no guide exists. Do you happen to have guidelines for the procedure?

CsepregiArtur, 2019-10-03:
I'm using my with esp3d
it is very simple or you can use octoprint, ok fot that you need a raspberry pi zero w

grego1981, 2019-10-03:
> I'm using my with esp3d
> it is very simple or you can use octoprint, ok fot that you need a raspberry pi zero w

Did you follow some guide  or how to? do you have any procedure so as to follow?
I already have octoprint connected but I have the need to sometimes use the printer without it, or when octprint  may be crashed or unavailable

rainerschulte, 2020-02-22:
Hi Folks!
Can I hook up here? 
I have a ESP01-S configured by ESP3D Instructions here in GitHub. 
To to this, I used the ESP-Programmer and the Firmware supplied by Luc. 
After Installation I got the Homepage and configured my configuration.h with enabeling WiFi and set Credentials.
After Uploading the Marlin Stuff again to the SKR-Board and plugged in the WiFi-Module in the Display I still don't get any Connection. Even the Message on Screen is not displayed. 
What else is missing?