bigtreetech/BIGTREETECH-SKR-V1.3 #91: skr v1.3 with tft35 v3 issue

brasco901, 2019-10-03:
I have my skr v1.3 set up to use neopixel led lights.  Everything works fine when using the TFT35 V3 screen in lcd12864 mode.  The led lights do not work in serial/touchscreen mode.  Any idea why it doesn't?

Kiethwallace, 2019-10-10:
Brasco -  Im new to the SKR v1.3 and tft35 - Im converting a davinci 1.0 with them. I can tell you that I have all but given up trying to make the touch mode work correctly. Even "out of the box" with their unmodified firmwares you cant issue more than a few commands before the skr stops responding. In addition, Despite a known working esp32, I can not get it to communicate with skr unless the screen is in touch mode - and it will do the same thing- allows a few commands and then skr freezes. I have no trouble with octoprint, ssd1306, or running in 12864 mode. Bottom line, I personally think the tft35 software is extremely buggy and is NOT going to work correctl until big tree updates it. I wish I had better news, im just sharing my 20+ hours trying to make it work. So for now, no esp8266, im using a wifi sd card and octoprint, and just leaving the display in 12864. That means I spend about 20 bucks too much, since I can get a 12864 display for less than 10 bucks.

Are you listening @bigtreetech ? We need firmware that is not pre-beta. This item was sold as a working solution for your skr v1.3 and it is anything but working... I wouldnt be upset if the ad for the tft35 said it was a beta, but thats not what we are led to believe.

Kiethwallace, 2019-10-10:
BTW, to better answer your question, the serial port used for your LED is (probably) also being used for the touch communication. Try changing the pin/port.

closed 2020-10-23 by brasco901