bigtreetech/BIGTREETECH-SKR-V1.3 #94: Let's talk about E0

Ph0rkoz, 2019-10-13:
I have LV8729 drivers and E0 doesn't move. The board boots up in reset and stays there until I remove the driver.

Rumblings in forums say its due to something the the SKR 1.3 bootloader. Is there a fix I can flash with jtag? There has been all this talk but never a fix.

zebolc, 2019-11-07:
Do you have enabled Linear advance?

Ph0rkoz, 2019-11-20:
no, its not a configuration issue!

Ph0rkoz, 2019-12-10:
Ok, finally I think there is a fix. Thanks to helpful people out there:
https://github.com/Arksine/LPC17xx-DFU-Bootloader/releases/tag/0.4.1

Kraken3d, 2020-02-28:
> Ok, finally I think there is a fix. Thanks to helpful people out there:
> https://github.com/Arksine/LPC17xx-DFU-Bootloader/releases/tag/0.4.1

I have the SKR 1.4 Turbo and when swapping the TMC 2209 to the LV8729 I get zero movement. What did you have to do to get it working or were you able to?