bigtreetech/BIGTREETECH-SKR-V1.3 #95: Stuttering motion, Display locking up

Rdog618, 2019-10-15:
Hey Guys,
This is driving me insane.  I have a CR-10 Mini that I am trying to get to run on this BIQU SKR 1.3 with 2208 drivers combo I got off amazon.  At first, I thought I had just received a bad motherboard, but after trying a new motherboard I have the exact same identical problem.  The board appears to operate fine when its not printing.  The menu navigates correctly and smoothly, I can make adjustments with no problem.  As soon as I begin a print, the menu becomes unresponsive and intermittent.  The motion of the printer is in a stuttering jerking motion, particularly when going around circles or corners.  When its printing a straight line I can occasinally get the menu to respond.  It will usually finish a print but occasionally freezes up and even if it does finish, the print looks awful because of the jerking and stuttering motion during the print.  It runs silent with stealth cop on, but when stealth chop is turned off there is an awful high pitched, high frequency ringing noise coming from the steppers.  Any help with this would be much apprechiated as this is a nightmare to trouble shoot.

Rdog618, 2019-10-26:
These photos do an amazing job capturing the studder effects.  I am suspicious the corrections may lie in either the planner or the way its calculating segments in arcs.  It doesn't seem to be hardware related.  I have been meaning to test some new config variables but a herniated disc has me behind lately.  Visually it seems like the planner is erroneously decelerating between segments in the arc.

DasGrinch, 2019-10-26:
Thanks.  So far I have tried (in no particular order) new stepper drivers, swapping stepper drivers between axises, swapping the lcd screen with encoder for a touchscreen and printing via the sd card slot on it, new microsd card, turned off tmc debug, turned off driver monitoring, turning current down to 400mA, up to 1000mA, turning off stealthchop and using spreadcycle (noisy, AND jerky).  I've used new gcode and old gcode.  Things I plan to try next include printing via USB (either through pronterface or via a raspberry pi).

DasGrinch, 2019-10-28:
I believe I have figured this issue out.  Someone on reddit advised me that I should try the newer version of the TMCStepper library.  I was using the latest Marlin build, but not the newest library for the stepper drivers.  I updated to v0.5.2 and re-compiled, flashed, and I have much better movement around curves and complicated edges.  I'd give it a try and see if it makes any difference for you.  To update I went into ProjectIO home, then libraries, then updates, and did it from there on my project.

Rdog618, 2019-10-28:
Interesting, I will have to try this as well.  

Apparently in the new version of marlin, Junction Deviation is enabled by default instead of classic Jerk control.  I was told that by disabling JD and reverting back to classic jerk may help and it certainly seems to have eliminated a significant amount of stutter.

Rdog618, 2019-10-28:
Despite the fact that I just downloaded cura 4.3 yesterday, it seems to have been the source of the problem.  I just downloaded Slic3r and loaded up my torture test of curvature and it is printing magnificently now!