bigtreetech/BIGTREETECH-SKR-V1.3 #99: 2130 spi not connecting

Ordagon, 2019-10-18:
Hi I recently bought a SKR V1.3 with 2130 from Bigtree. I want to run then 2130 in SPI, I have edited the firmware following the guides, set the jumpers on the board and done the soldering on the stepper drivers as the instructions require. But there is a problem and the drivers do not connect to the board

What can I do?

Connecting...
Printer is now online.
>>>m122
SENDING:M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	750	800	850	850
RMS current	734	795	826	826
MAX current	1035	1121	1165	1165
Run current	23/31	25/31	26/31	26/31
Hold current	11/31	12/31	13/31	13/31
CS actual	31/31	31/31	31/31	31/31
PWM scale	255	255	255	255
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		0	0	0	0
tstep		4294967295	4294967295	4294967295	4294967295
pwm
threshold
[mm/s]
OT prewarn	true	true	true	true
off time	15	15	15	15
blank time	54	54	54	54
hysteresis
-end		12	12	12	12
-start		8	8	8	8
Stallguard thrs	8	8	0	0
DRVSTATUS	X	Y	Z	E
stallguard		*	*	*	*
sg_result	1023	1023	1023	1023
fsactive		*	*	*	*
stst		*	*	*	*
olb		*	*	*	*
ola		*	*	*	*
s2gb		*	*	*	*
s2ga		*	*	*	*
otpw		*	*	*	*
ot		*	*	*	*
Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
		Y	0xFF:FF:FF:FF	 Bad response!
		Z	0xFF:FF:FF:FF	 Bad response!
		E	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH
Testing Z connection... Error: All HIGH
Testing E connection... Error: All HIGH
ok P15 B3

Ordagon, 2019-10-19:
all right found the solution. I was testing the board via USB power, this seems to not have enough power to turn on the drivers. Tested with the power supply everything works as it should.

closed 2019-10-19 by Ordagon

Talha909, 2019-12-17:
> all right found the solution. I was testing the board via USB power, this seems to not have enough power to turn on the drivers. Tested with the power supply everything works as it should.

Can you please share the config file and config file adv with me.