bigtreetech/BIGTREETECH-SKR-V1.3 #101: Skr 1.3 driver errror  Bad response!

trunglp, 2019-10-19:
Hi everyone,
Today i problem with board BIGTREETECH-SKR-V1.3

i change or replace new TMC2208 but not work. 

how to fix it ? 

 
M122
SENDING:M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	580	580	580	650
RMS current	550	994	550	642
MAX current	776	1402	776	905
Run current	17/31	17/31	17/31	20/31
Hold current	8/31	8/31	8/31	10/31
CS actual	8/31	0/31	8/31	10/31
PWM scale	10	0	10	12
vsense		1=.18	0=.325	1=.18	1=.18
stealthChop	true	false	true	true
msteps		16	256	16	16
tstep		max	0	max	max
pwm
threshold
[mm/s]
OT prewarn	false	false	false	false
off time	4	0	4	4
blank time	24	16	24	24
hysteresis
-end		2	-3	2	2
-start		1	1	1	1
Stallguard thrs
DRVSTATUS	X	Y	Z	E
stst		*		*	*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:08:00:00
		Y	0x00:00:00:00	 Bad response!
		Z	0xC0:08:00:00
		E	0xC0:0A:00:00
Testing X connection... OK
Testing Y connection... Error: All LOW
Testing Z connection... OK
Testing E connection... OK

trunglp, 2019-10-23:
I don’t jumper usb, it only error axis Y, you can see my video

fiferboy, 2019-10-24:
I have the exact same problem with my current board. Y axis shows the same error, all the other axis are fine in UART mode. I am using the V3.0 TMC2208 from BBT that are already shipped with UART enabled.

My experience is virtually identical to that in issue #75 (and I outlined the steps I already took there) so I am going to try a replacement board. In the meantime I run Y axis in DIR (non-UART) mode in order to have a functioning board.

hiewliwen, 2019-12-05:
> I have the exact same problem with my current board. Y axis shows the same error, all the other axis are fine in UART mode. I am using the V3.0 TMC2208 from BBT that are already shipped with UART [...]

I am having the same the issue. Do I just remove the jumper for the SPI on that particular axis to get it in non-UART mode?

thomasdr1, 2019-12-07:
I have the Bigtreetech TMC2208 V3 which should work in UART mode.
The jumpers and Marlin 2 are set according to instructions.
But not a single TMC2208 works in UART mode.

kasimsuzen, 2020-01-16:
> I have the Bigtreetech TMC2208 V3 which should work in UART mode.
> The jumpers and Marlin 2 are set according to instructions.
> But not a single TMC2208 works in UART mode.

i have the same issue did you find a solution?

hiewliwen, 2020-01-20:
> > I have the Bigtreetech TMC2208 V3 which should work in UART mode.
> > The jumpers and Marlin 2 are set according to instructions.
> > But not a single TMC2208 works in UART mode.
> 
> i have the same issue did you find a solution?

I suggest getting BTT to do an exchange for you. 
I had 1 out of 4 2209 not working on UART. They sent a new piece after I submit video evidence of the error.

freq85hz, 2021-02-01:
Hi guys. I had this problem too. Always TMC connection error. I have 2208 3.0 from two trees. Its only nesesery to solder. For now its work without problems. When you look at your TMC from under side. screw is on the down. On left side are 3 soldered connections ( around on the middel)
You have to connect 2 of them. First two from down side. Nothing more.