bigtreetech/BIGTREETECH-SKR-V1.3 #103: Not showing print progress and percentage in the display (Bigtreetech TFT 35 V2.0)

KrishnaMohanAV, 2019-10-20:
I am able to print from onboard sd card in SKR V1.3, but printing details are not showing in the TFT display. Please help.

ravecaron, 2019-10-22:
How did you get the TFT to connect to the skr v1.3 in the first place?

ravecaron, 2019-10-22:
I have been having a lot of problems with it... it turns on but says no printer is connected

KrishnaMohanAV, 2019-10-23:
I Enabled #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER in configuration.h and  #define SDCARD_CONNECTION ONBOARD in configuration_adv.h , it worked for me.

ipatch, 2019-11-03:
@KrishnaMohanAV not trying to highjack your issue,

@ravecaron 

I just received the tft35 v2 in the mail yesterday, the first thing i'd do is build and install the latest firmware from this git repo on your screen.

when i check the info of my screen it is

```
Firmware: TFT35_V2.0.24.1 Nov 2 2019
```

The **date** will be the date in which you built the FW for the screen.

After you've updated the firmware for the screen, then you'll need to make certain your `Configuration.h` and accompanying marlin configuration files are configured properly, as I to was running into issues with the screen displaying the,

```
no printer attached to screen message
```

on my LCD.

The settings for marlin that worked for me are,

**Configuration.h**

```c
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 115200
```

Initially I had set the baudrate to invalid baudrate thus the lcd was unable to properly communicate with the board, you can verify the connection by sending **m** commands for the lcd itself.

rydol, 2020-03-25:
Hi @KrishnaMohanAV  I just came to the same problem, since the slot on my TFT V2.0 cant read the SD card anymore I switched to an onboar sd card ond SKR e3 mini. It works, but it does not show the progress while printing. Did you manage to solve the issue ??

KrishnaMohanAV, 2020-03-28:
> Hi @KrishnaMohanAV I just came to the same problem, since the slot on my TFT V2.0 cant read the SD card anymore I switched to an onboar sd card ond SKR e3 mini. It works, but it does not show the progress while printing. Did you manage to solve the issue ??

Hi @rydol Problem still exists.

KrishnaMohanAV, 2021-02-17:
Issue solved by flashing frimware : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Bootloaders/TFT35_V2%20Bootloader%20fix

closed 2021-02-17 by KrishnaMohanAV