bigtreetech/BIGTREETECH-SKR-V1.3 #107: cant connect to the board.

smintjes, 2019-10-24:
I flashed the firmware ( stock & config modified for my needs ), and everything seems to go well. I got serial mapped to 0, and serial 2 to -1 ( for usb ) . However, im not able to connect to the board using any 3d printer software. The board is recognized as USB TO SERIAL (COM 10). 
If i'm right, it should be detected as Marlin.

Any clue what is going wrong here?

smintjes, 2019-10-24:
What i tried: 
- switching SD cards
- trying the newer marlin firmware ( not ny bigtreetech )
- trying stock images

Nothing works. Maybe the board is just faulty?

ipatch, 2019-11-03:
I'm going to assume that your using some version of Windows due to the fact that you mention **COM 10**, and I believe that is a windows***ism***, as on *nix and macOS  the board if configured properly shows up as either a `/dev/tty.` device or a `/dev/tty.cu` device depending on how you configured your serial ports within your `Configuration.h`.

For the current [marlin fork/branch](https://github.com/ipatch/Marlin/tree/ipatch/build-marlin-2.x) that I've been working with I have the below [serial port configuration setup](https://github.com/ipatch/Marlin/blob/ipatch/build-marlin-2.x/Marlin/Configuration.h#L121) in my `configuration.h`

```c
#define SERIAL_PORT -1 // default
// ipatch `SERIAL_PORT -1` refers to the serial port builtin to the physical USB port on the brd.
// #define SERIAL_PORT -1 // usuall req for connecting via prontoface
// #define SERIAL_PORT_2 -1 // TODO
#define SERIAL_PORT_2 0 // SKR REF: https://youtu.be/GNGN2iSQ5j4?t=515 
// #define BAUDRATE 250000 // default // TODO
#define BAUDRATE 115200
```

> ❗️ I have the bigtreetech tft35 v2 display which requires a baudrate of 115200 or the display will not recognize the printer.

With those above listed serial port configuration lines, 
1. i compile marlin 2.0.x bugfix branch with my modifications added to the configuration files
2. plugin the SD card I use to upload the `firmware.bin` file to the SD card attached to my computer
> in the `platform.ini` file there is a setting that can be added when compiling marlin with PlatformIO to specify a disk to copy the compiled `.bin` file to.  I formatted the 128MB uSD card that came with the board using fat32 filesystem with a MBR partition map. and set the volume label to **SKR**, and within the `platformio.ini file i have added the following line,

```
[env:LPC1768]
# protocol          =  mbed
upload_port       = /Volumes/SKR
```

add the **upload_port** line within the **[env:LPC1768]** if you'd like platformio to auto copy the built bin file to the sd card.

3. Insert the sd card in the board, make certain the jumper on the board is configured with 5V USB power, it's the little red jumper close to the USB port and should be attached to the two right pins in order for the board to powered via USB, and if all things go well, you should see a little red LED light up on the board letting you know the board has power.

4. So with the SD card inserted into the uSD port of the board press the reset button on the board give a few seconds and the board will copy the newly compiled firmware to the board, and you should be running the version of marlin you compiled on your board.

5. you can verify all this by connecting your computer which more than likely is already connected to the board because that is how you are powering the board.

6. Open printrun ie. prontoface and verify the firmware on the board
> obviously printrun / prontoface will need to be installed on your computer to run the program.

7. once printrun is open select the port you'd like to connect to, specify the baudrate to use, ie. **115200**
> on macos, the device is listed as `/dev/tty.usbmodem14111` obviously windows will display something different, ie. com port 10

8. verify your firmware is running on your board

  a. within the right side of prontoface input the following **m** command

```
m115
```

👆 that command should show useful information about the firmware running on the board.

9. you can remove the SD card without having to reboot the board if you'd like.

> ❗️❗️❗️ I'd set aside that particular sd card and use it just for updating the board firmware and not rename or use it for anything else.

Below is a picture what the output of the `m115` command looks like on my box, ie. macOS

[image]

##

I even was able to install printrun on a windows 10 vm i maintain using virtualbox, and below are some pictures of me connecting to the device using prontoface on windows 10

[image]

##

[image]

##

[image]

##

> ❗️❗️❗️the **com** port number will more likely differ based upon how your system is configured.