bigtreetech/BIGTREETECH-SKR-V1.3 #109: E0 and E1 not working/responding

krizo96, 2019-10-27:
This problem is almost certainly the result of my error, but I want to see if there's any way to remedy it short of replacing the board.
What happened is that I accidentally shorted the Z end stop and my E0 stepper driver and port died. Along with that I also found out that the pins for X and Y end stops didn't work anymore, but I worked around all of that by remapping the pins and the end stops work now, however, I still can't get my E0 or E1 to work no matter what I do.
I tested the drivers and motors I'm using and they all work fine on X, Y and Z axis, but on E0 the motor locks (probably because of the damage) and when I try to remap so I use E1 as E0 it's completely unresponsive - no current, no nothing.
Any ideas on how to get my E1 to work? And in case it's dead, is there any way to get the 4th axis/extruder working besides getting a new board?

Thanks!

zebolc, 2019-11-07:
Do you have enabled Linear advance? Try to disable it. Or try playing with MINIMUM_STEPPER_DIR_DELAY, MINIMUM_STEPPER_PULSE and MAXIMUM_STEPPER_RATE. I'm using 20, 2, 400000 with linear advance and it solved my issue with E0 not working.

krizo96, 2019-11-08:
> Do you have enabled Linear advance? Try to disable it. Or try playing with MINIMUM_STEPPER_DIR_DELAY, MINIMUM_STEPPER_PULSE and MAXIMUM_STEPPER_RATE. I'm using 20, 2, 400000 with linear advance and it solved my issue with E0 not working.

I didn't have Linear advance or any of the settings you mentioned enabled, but playing around with them didn't help. Still not working.

mr-september, 2024-06-02:
Did you get anywhere with this? Facing the same issue.