bigtreetech/BIGTREETECH-SKR-V1.3 #110: New SKR 1.3 build with TMC2130 drivers - Steppers Not Working

Redssnake77, 2019-10-27:
Hi,

I have just finished my upgrade of a Geeetech i3 Pro B with an SKR 1.3 with TMC2130 stepper drivers. The hotend works, and the heatbed works, but I can't get any of the stepper motors to move. I have set the vref on each of the drivers to a close to 800mv as I could get them which matches the software. Comms work fine, I can increase and decrease the vref via the LCD, but it seems that the drivers wont 'enable'. I've tried using M17, but the report when using M122 is that they are disabled. This is a copy of the report:

Send: M122 Recv: X Y Z Z2 E
Recv: Enabled false false false false false
Recv: Set current 800 800 800 800 800
Recv: RMS current 795 795 795 795 795
Recv: MAX current 1121 1121 1121 1121 1121
Recv: Run current 25/31 25/31 25/31 25/31 25/31
Recv: Hold current 12/31 12/31 12/31 12/31 12/31
Recv: CS actual 12/31 12/31 12/31 12/31 12/31
Recv: PWM scale 0 0 0 0 0
Recv: vsense 1=.18 1=.18 1=.18 1=.18 1=.18
Recv: stealthChop true true true true false
Recv: msteps 16 16 16 16 16
Recv: tstep max max max max max
Recv: pwm
Recv: threshold 101 101 658 658 31
Recv: [mm/s] 100 100 3 3 30
Recv: OT prewarn false false false false false
Recv: OT prewarn has Recv: been triggered false false false false false
Recv: off time 3 3 3 3 3
Recv: blank time 24 24 24 24 24
Recv: hysteresis
Recv: -end -1 -1 -1 -1 -1
Recv: -start 1 1 1 1 1
Recv: Stallguard thrs 0 0 0 0 0
Recv: DRVSTATUS X Y Z Z2 E
Recv: stallguard Recv: sg_result 0 0 0 0 0
Recv: fsactive Recv: stst * * * * *
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers: Recv: X 0x80:0C:00:00 Recv: Y 0x80:0C:00:00 Recv: Z 0x80:0C:00:00 Recv: Z2 0x80:0C:00:00 Recv: E 0x80:0C:00:00 Recv:
Recv:
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: ok

I have tried increasing the vref, and tried to double check the stepper motor connections which seem to match the SKR.

I'm at complete loss. Any help would be appreciated.

anggiatm, 2019-11-16:
what jumpers configuration on board? can you send the pic

turbojunke, 2020-01-21:
Any update on this?? I have basically the same issue except with tmc2209 drivers... My stepper motors won't move at all.

Licokite, 2021-01-05:
Hi,

Same board with TMC2130, same issue. Any fix ?!