bigtreetech/BIGTREETECH-SKR-V1.3 #115: Print freezes but still have access to lcd screen menu.

Loadie150, 2019-11-01:
I’ve had the Skr v1.3 for a few months now and it’s worked flawlessly until recently. I’ve burnt out 2 touch sensors and now when in the middle of a print, over an hour or so long, well, the print just stops! I have full access to the lcd screen and menus. I can select pause and then resume, nothing. I can move the axis and back, no continuations of the print. I always have to hard boot and start over again. I’ve wasted about a spool of filament on this issue. Yesterday I updated to the newest firmware here. Same results, only a little sooner.

Core A8 XY slim printer with
Skr v1.3
TMC2208s in UART mode
NPN sensor
Dual z axis
Dual PSUs.

Any help would be appreciated!

anggiatm, 2019-11-16:
i have same issue,

in marlin configuration.h
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

if you print via lcd you sould unplug the usb cable from pc

Loadie150, 2019-11-16:
I’ve been dealing with Bigtreetech and they’re sending me a replacement board. I also have another 12864 lcd coming as well. Will figure it out sooner or later.

My Marlin settings are the same.