bigtreetech/BIGTREETECH-SKR-V1.3 #116: Can someone put him config files with TMC2130 SPI v3.0

nicstoun, 2019-11-02:
I want some help me  with same setup attach him  config files there ...   because  i move these  drivers from E3 DIP to SKR 1.3 with same Marlin , changed only address for  drivers on this board , but cannot get these drivers work on SKR . All jumpers  its same place like on picture there for TMC2130 SPI . i not have idea now  why  board not see it.  Attach my config files .

[Configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3800436/Configuration_adv.zip)

Thanh-Viet, 2019-11-03:
I have a SKR1.3 TMC2130 BLTouch Ender 3 Pro if you want. 

https://github.com/Thanh-Viet/Marlin/tree/SKR1.3_TMC2130v30_BLTouch?files=1

nicstoun, 2019-11-05:
Thank you i try ,, i almost   try everything, yesterday     start to talk with bigtreetech  , and they send me some  firmware's again  without success ,   they now think  for replace board.  Not good  ..why bigtreetech dont  made  simple test before sold these boards.

Talha909, 2019-12-11:
> I have a SKR1.3 TMC2130 BLTouch Ender 3 Pro if you want. https://github.com/Thanh-Viet/Marlin/tree/SKR1.3_TMC2130v30_BLTouch?files=1
> […](#)
> -- A bientôt, Best Regards, Bertrand Jenner

Please send me the config and config adv file

Talha909, 2019-12-11:
> I have a SKR1.3 TMC2130 BLTouch Ender 3 Pro if you want. https://github.com/Thanh-Viet/Marlin/tree/SKR1.3_TMC2130v30_BLTouch?files=1
> […](#)
> -- A bientôt, Best Regards, Bertrand Jenner

can you send me the config and config adv file