bigtreetech/BIGTREETECH-SKR-V1.3 #118: Stuck at "Thermal Runaway: E1"

andrebianchessi, 2019-11-04:
Everytime I reset, the screen says:
"Thermal runaway: E1
Printer Halted
Please Reset"

This keeps happening even just the lcd (Exp1 and Exp2) are attached.

Besides, my computer now can't recognize the board.

AnHardt, 2019-11-04:
Likely this happens **because** you have temperature sensors defined and compiled in, but no sensors or heaters connected.
As long as no real sensors are connected try 999 and/or 998.

andrebianchessi, 2019-11-18:
I attached the sensors and heaters correctly, and set:
#define BOGUS_TEMPERATURE_GRACE_PERIOD 60000
so that the temperature peak would be ignored for 1 minute to avoid maxtemp error.

The hotbed sensor is reading the correct temperature (25°C), but the extruder displays a fixed temperature of 300°C, even if I heat it with a lighter from a distance.
I measured the extruders thermistor resistance (disconected from the circuit) and got a value of around 100k that decreases with temperature.

I tried connecting the extruders thermistor to the bed thermistor input on the board (TB), and I got a correct measurement on the "bed temperature" (which is actually the extruders temperature). However, if I connect the bed thermistor to the extruders thermistor input on the board (TH0), the "extruders temperature" (which is actually the bed temperature) is still at 300°C.

The thermal setting in configurations.h are the following:
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_5 0
#define TEMP_SENSOR_BED 1
#define TEMP_SENSOR_CHAMBER 0

I can't find a solution to this. Any help is highly appreciated.

burakerenn, 2019-12-28:
Did you find out anything ? I am getting same problems as well in both of my boards

edwilliams16, 2020-08-02:
@andrebianchessi @burakerenn   Any resolution?  I have two boards with the same problem.  In each case when I measure the resistance between the TH0 pins I see a short circuit, TH1 and TB measure 6.2K.  Do you have the same?

burakerenn, 2020-08-02:
Hi, yes I solved the problem. The issue is same , pins are short. That means pins on MCU is also shorted. I examine the datasheet. All in all, contact your seller and tell that your board is broken. No way to fix it.