bigtreetech/BIGTREETECH-SKR-V1.3 #119: DRV8825 not working on SKR1.3 for 1 driver(E0)

legojedidev, 2019-11-04:
So I use an Anet a8 but recently my board caught on fire. I decided to upgrade to an SKR 1.3 and bought 5 drv8825 stepper drivers. Everything works fine except for my E0(E1 is being used for my other z-axis). My E0 motor gets current, but it barely moves. Any idea why this may be happening? I am running Marlin 2.0 and can upload my configuration files later. Also I can't get my LCD 2004 to work. Any help is appreciated!

Tomthi, 2019-11-06:
Have you tried adjusting the vREF for that stepper driver? They may be set too low. I had issues with the voltages and after adjusting them performance was much better.

legojedidev, 2019-11-06:
I just figured out that the driver itself was causing a short

closed 2019-11-06 by legojedidev

erberru70, 2019-11-29:
> Así que uso un Anet a8 pero recientemente mi placa se incendió. Decidí actualizarme a un SKR 1.3 y compré 5 controladores paso a paso drv8825. Todo funciona bien, excepto mi E0 (E1 se está utilizando para mi otro eje z). Mi motor E0 recibe corriente, pero apenas se mueve. ¿Alguna idea de por qué esto puede estar sucediendo? Estoy ejecutando Marlin 2.0 y puedo cargar mis archivos de configuración más tarde. Además, no puedo hacer que mi LCD 2004 funcione. Cualquier ayuda es apreciada!

hola , como conseguiste cargar marlin 2.0 en la placa skr v1.3, por que a mi no para de darme error al compilar marlin con vscode