bigtreetech/BIGTREETECH-SKR-V1.3 #121: pin configuration and marlin 2.0

ipatch, 2019-11-06:
I've slowly been working towards installing a skr v1.3 for my cr10s pro, and am running into some questions regarding the cooling fans.

- Is there no pin assignment number that I can configure for the pins I outlined with the purple circle below?  Currently I have my "always on hot end" fan hooked up to the pins outlined in the purple circle.

- Also, I'd like to wire up the part cooling fan to pin 2.3, and when I test out the functionality using the tft35 v2 i see a little green led light up next to one of the mosfets, but when my multi meter is hooked up to the pins i do not get any voltage coming from pin 2.3 when i set the fan speed to 255 or 0.  however the green led does turn off and on with switching from 0 to 255 from the lcd menu.

Tomthi, 2019-11-06:
There is no pin assignment for the 12/24v always on. 
In my configuration, I have my part cooling fan connected to P2.3 and my hot end cooling fans are wired up to MOSFETs and triggered off the unused end stop pins. This works well, the hot end fans come on when the hotend reaches 50c. But according to the docs, if you only have one extruder your hotend fan should be connected to HE1 P2.4. Your part cooling fan should work find on p2.3

As defined in pins.h

> // Heaters / Fans
//
#define HEATER_0_PIN       P2_07
#if HOTENDS == 1
  #define FAN1_PIN         P2_04
#else
  #define HEATER_1_PIN     P2_04
#endif
#define FAN_PIN            P2_03
#define HEATER_BED_PIN     P2_05

ipatch, 2019-11-06:
> As defined in pins.h
> 
> > // Heaters / Fans
> > //
> > #define HEATER_0_PIN       P2_07
> > #if HOTENDS == 1
> > #define FAN1_PIN         P2_04
> > #else
> > #define HEATER_1_PIN     P2_04
> > #endif
> > #define FAN_PIN            P2_03
> > #define HEATER_BED_PIN     P2_05

I don't see those configuration lines specifically within the `pins.h` source code file for the marlin 2.0 source code but rather within the `pins_BIGTREE_SKR_V1.3.h` file. That said I have the following code block within the `pins_BIGTREE_SKR_V1.3h` file.

```
//
// Heaters / Fans
//
#ifndef HEATER_0_PIN
  #define HEATER_0_PIN     P2_07
#endif
#if HOTENDS == 1
  #ifndef FAN1_PIN
    #define FAN1_PIN       P2_04
  #endif
#else
  #ifndef HEATER_1_PIN
    #define HEATER_1_PIN   P2_04
  #endif
#endif
#ifndef FAN_PIN
  #define FAN_PIN          P2_03
  // #define FAN_PIN P2_04
#endif
#ifndef HEATER_BED_PIN
  #define HEATER_BED_PIN   P2_05
#endif
```

And link to a video that I just recorded to further explain what's going on.  I hope it's just a misconfiguration on my part, and not a faulty board.  Thanks for your comment BTW.

Tomthi, 2019-11-06:
Are you running your tests from the USB power or Power supply ?

Well you should not need to mess with the pins file.  In the case of my part cooling fan, since I have it hooked up to to the CNC fan pin P2.3 I can control it from the display, Temp/FAN Speed

The two pins goto MOSFETS.

> /**
 * Extruder cooling fans
 *
 * Extruder auto fans automatically turn on when their extruders'
 * temperatures go above EXTRUDER_AUTO_FAN_TEMPERATURE.
 *
 * Your board's pins file specifies the recommended pins. Override those here
 * or set to -1 to disable completely.
 *
 * Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
 */
#define E0_AUTO_FAN_PIN  P1_26 //-1
#define E1_AUTO_FAN_PIN  P1_28 //-1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1

ipatch, 2019-11-07:
@Tomthi 

https://www.youtube.com/watch?v=oD1DJbHIX3o

Tomthi, 2019-11-07:
I think you're in the wrong pins file. You should be referencing 

 Marlin/Marlin-bugfix-2.0.x/Marlin/src/pins/lpc1768/pins_BIGTREE_SKR_V1.3.h

ipatch, 2019-11-07:
@Tomthi 

thanks for your assistance, FWR I had some weird if guards in my pins configuration file, not sure if was from pulling in some upstream changes from the marlin bugfix branch the other day, but needless to say i reviewed the pins configuration file within this repo specifically for the SKR v1.3 board, and updated my specific build branch of marlin to reflect the changes in this repo, and now things are working as should and my meter is reporting the proper voltage.

w00t ✧*｡٩(ˊᗜˋ*)و✧*｡

##

For future me, [see](https://github.com/ipatch/Marlin/blob/c01e03a58bebc572f45ab38bd2f27874e39f3d31/Marlin/src/pins/lpc1768/pins_BIGTREE_SKR_V1.3.h#L180)

closed 2019-11-07 by ipatch

scorninpc, 2021-03-28:
@ipatch

Page not found =/

https://github.com/ipatch/Marlin/blob/c01e03a58bebc572f45ab38bd2f27874e39f3d31/Marlin/src/pins/lpc1768/pins_BIGTREE_SKR_V1.3.h#L180