bigtreetech/BIGTREETECH-SKR-V1.3 #122: TMC Connection error and ALL LOW - TMC2209

Djinn1, 2019-11-07:
Hi,

I having problem with Fysetc tmc2209 and SKR 1.3. Iam getting ALL LOW when running M122 and in display it says TMC Connection error.

I have tried to move the axis and it seems all work.

The power supply is conncted and the pins is INT +5V.

Have serched for hours and cant find a fix. Need help.

See picture.

Send: M122
Recv: 		X	Y	Z	E
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false
Recv: Set current	650	650	580	650
Recv: RMS current	1160	1160	994	1160
Recv: MAX current	1636	1636	1402	1636
Recv: Run current	20/31	20/31	17/31	20/31
Recv: Hold current	10/31	10/31	8/31	10/31
Recv: CS actual	0/31	0/31	0/31	0/31
Recv: PWM scale	0	0	0	0
Recv: vsense		0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false
Recv: msteps		256	256	256	256
Recv: tstep		0	0	0	0
Recv: pwm
Recv: threshold
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time	0	0	0	0
Recv: blank time	16	16	16	16
Recv: hysteresis
Recv: -end		-3	-3	-3	-3
Recv: -start		1	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW

[image]

Djinn1, 2019-11-10:
Problem was on this brand there is a RX TX on top you should put a jumper on. Now everything works.

closed 2019-11-10 by Djinn1

Regaor, 2020-03-01:
hello dear friend Djinn1,
I have the same problem with MKS S-Gen-L and Fysetc TMC2209 V3.0.
Could you help me please.
Sorry for my English, I am French.
Thanks a lot
Regis