bigtreetech/BIGTREETECH-SKR-V1.3 #123: Cant build marlin 2.0: .pio\build\LPC1768\src\src\Marlin.cpp.o] Error 1

EJtechandDiy, 2019-11-07:
Everytime I try to build I get this error:

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; framework: arduino; board: nxp_lpc1768)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html

PLATFORM: NXP Arduino LPC176x 0.1.0 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.80KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.1
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.5.2
|-- <Adafruit NeoPixel> 1.2.4
|-- <SailfishLCD>
Building in release mode

Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmwa
re.bin) manually to the appropriate disk
 
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\HAL_LCD_I2C_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\HAL_LCD_pin_routines.c.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Marlin\src\HAL\HAL_LPC1768\HAL.cpp: In function 'int16_t PARSED_PIN_INDEX(char, int16_t)':
Marlin\src\HAL\HAL_LPC1768\HAL.cpp:60:73: error: 'GET_PIN_MAP_INDEX' was not declared in this scope
   const  int16_t ind = (port < ((NUM_DIGITAL_PINS) >> 5) && pin < 32) ? GET_PIN_MAP_INDEX((port << 5) | pin) : -2;
                                                                     Marlin\src\HAL\HAL_LPC1768\HAL.cpp: In function 'int16_t PARSED_PIN_INDEX(char, int16_t)':
Marlin\src\HAL\HAL_LPC1768\HAL.cpp:60:73: error: 'GET_PIN_MAP_INDEX' was not declared in this scope
   const  int16_t ind = (port < ((NUM_DIGITAL_PINS) >> 5) && pin < 32) ? GET_PIN_MAP_INDEX((port << 5) | pin) : -2;
                                                                         ^~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp: In function 'void spiBegin()':
Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:128:21: error: 'LPC1768_PIN_PIN' was not declared in this scope
     PinCfg.Pinnum = LPC1768_PIN_PIN(SCK_PIN);
                     ^~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:128:21: note: suggested alternative: 'LPC_PINCON'
     PinCfg.Pinnum = LPC1768_PIN_PIN(SCK_PIN);
                     ^~~~~~~~~~~~~~~
                     LPC_PINCON
Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:129:22: error: 'LPC1768_PIN_PORT' was not declared in this scope
     PinCfg.Portnum = LPC1768_PIN_PORT(SCK_PIN);
                      ^~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:129:22: note: suggested alternative: 'LPC176X_FRAMEWORK'
     PinCfg.Portnum = LPC1768_PIN_PORT(SCK_PIN);
                      ^~~~~~~~~~~~~~~~
                      LPC176X_FRAMEWORK

In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/./HAL_LPC1768/HAL.h:42,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/../HAL/HAL.h:26,
[... 162 lines cut ...]
Marlin\src\HAL\HAL_LPC1768\../../sd/../inc/../HAL/./HAL_LPC1768/fastio.h:105:35: note: in expansion of macro '_WRITE'
 #define SET_OUTPUT(IO)        do{ _WRITE(IO, _READ(IO)); _SET_OUTPUT(IO); }while(0)
#define _READ(IO)             READ_PIN(IO)
                               ^~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../../sd/../inc/../HAL/./HAL_LPC1768/fastio.h:105:46: note: in expansion of macro '_READ'
 #define SET_OUTPUT(IO)        do{ _WRITE(IO, _READ(IO)); _SET_OUTPUT(IO); }while(0)

                                              ^~~~~
Marlin\src\HAL\HAL_LPC1768\../../sd/../inc/../HAL/./HAL_LPC1768/fastio.h:115:35: note: in expansion of macro 'SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\main.cpp:94:5: note: in expansion of macro 'OUT_WRITE'
     OUT_WRITE(SS_PIN, HIGH);
     ^~~~~~~~~
In file included from C:\Users\Eugenio\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21,
                 from C:\Users\Eugenio\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from C:\Users\Eugenio\.platformio\packages\framework-arduino-lpc176x\cores\arduino/CDCSerial.h:34,
                 from Marlin\src\HAL\HAL_LPC1768\main.cpp:31:
C:\Users\Eugenio\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:48:47: note:   'LPC176x::gpio_get'
 [[gnu::always_inline, nodiscard]] inline bool gpio_get(const pin_t pin) {
                                               ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o] Error 1

I hoop any one can help me with this problem.

komar007, 2019-11-10:
I've had what looks exactly like your problem using stock Marlin-2.
My reproduction was:

1. pull the latest upstream bugfix-2.0.x branch
2. rebase my changes on top of it
3. compile (works fine) 
4. go back a couple tens of commit earlier (because my printer was useless on HEAD) and rebase the same changes
5. compile - and everything broken.

I finally fixed it by removing the whole .pio directory and also ~/.platformio to force a platformio reinstallation in vscode. Magic...

mentaldemise, 2020-05-14:
I had this happen pulling in marlin changes because I didn't update plaform.ini along with the marlin source. Different versions in it than the one I had.

mentaldemise, 2020-05-14:
An addendum: I faced other errors in just trying to copy the src directory over-top of the one I had. Extra files in there were still trying to build and messing stuff up. So make sure you update platform.ino and *replace* the src folder if you're going that route.

jel111, 2021-01-04:
Had this happen to me. I had changed
` #define SERIAL_PORT 0 to #define SERIAL_PORT 2` 
changed it back to 0 and things compiled.