bigtreetech/BIGTREETECH-SKR-V1.3 #124: Junction deviation settings do not persist on SKR 1.3

appideasDOTcom, 2019-11-08:
I have been thinking this is a bug in Marlin, but doing what I am trying to do doesn't require a lot of options for it to work. I've already posted this question here:
https://reprap.org/forum/read.php?306,861601,861604#msg-861604
and here:
https://3dprinting.stackexchange.com/questions/11261/setting-junction-deviation-in-firmware-has-no-effect

Without repeating everything from those threads, I have set #define JUNCTION_DEVIATION_MM 0.02 in Marlin's Configuration.h and CLASSIC_JERK is disabled, but every time I reset or power cycle the SKR controller, the "Junction Dev" setting under Configuration > Advanced Settings > Jerk is set to 0.00, and the motors come to a full stop on every angle change.

Is it possible that this is a bug in the SKR 1.3? Does anyone have any ideas about what I can do to troubleshoot.

thisiskeithb, 2019-11-08:
Did you run `M502` and `M500` after updating your firmware?

appideasDOTcom, 2019-11-08:
> Did you run `M502` and `M500` after updating your firmware?

Yes. M502 resulted in "Hardcoded Default Settings Loaded" but no changes to behavior.

This is the third time that's been suggested, so I've tried it three or more times. Is this really necessary after compiling a new firmware and power cycling the controller? All other changes I made in the firmware have taken effect after compiling, loading and power cycling.

appideasDOTcom, 2019-11-08:
> Did you run `M502` and `M500` after updating your firmware?

I dug into this a little more, and I am either running into a bug, or I'm simply not understanding what I'm supposed to be seeing.

I send this command:
M205 J0.020
Then verify that junction deviation is set correctly (it is)
Then this:
M500
And that responds with:
Settings Stored (616 bytes; crc 51371)
Then I send this:
M503
And the report for M205 is missing the setting I just confirmed to have changed and been working:
...
M205 B20000.00 S0.00 T0.00
...
And power cycling or resetting the controller leads to the same result: Junction Deviation is reset to 0.

Shouldn't I see "J0.020" in the M205 line immediately after changing the setting, and verifying that it works?

appideasDOTcom, 2020-02-10:
Shoot. 

This was a bug between SKR boards and Marlin, but it has been resolved in a recent update on the Marlin Bugfix-2.0.x branch.

I forgot I opened this. Sorry for leaving it open for so long after the resolution.

closed 2020-02-10 by appideasDOTcom