bigtreetech/BIGTREETECH-SKR-V1.3 #125: Platform IO compiling error

sthgardner, 2019-11-11:
Stll can't compile or flash to SKR V1.3.

> Executing task: C:\Users\sthga\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.1.zip; framework: arduino; board: nxp_lpc1768)
--------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.5.2
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (a bytes-like object is required, not 'str')
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pioenvs/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp: Permission denied
compilation terminated.
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp: Permission denied
compilation terminated.
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\HAL.cpp: Permission denied
compilation terminated.
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\HAL_SPI.cpp: Permission denied
compilation terminated.
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\HAL_timers.cpp: Permission denied
compilation terminated.
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o] Error 1
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\MarlinSerial.cpp: Permission denied
compilation terminated.
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o] Error 1
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\fast_pwm.cpp: Permission denied
compilation terminated.
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o] Error 1
cc1.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c: Permission denied
compilation terminated.
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o] Error 1
================================================== [FAILED] Took 5.42 seconds ==================================================

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  FAILED    00:00:05.418
LPC1769                  IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      IGNORED
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
STM32F103CB_malyan       IGNORED
chitu_f103               IGNORED
teensy35                 IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
============================================ 1 failed, 0 succeeded in 00:00:05.418 ============================================ 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3831909/Marlin.zip)

hoyohayo, 2019-11-13:
I am encountering the same problem.
I tried fixing it in the platformio.ini.
There are a few things making this qute difficult. So first I got a vanilla marlin from the github branch, and made changes for my printer in the config_adv.h and config.h. Since I am using TMC2208 v3.0 in UART mode, I just followed the instructions by placing 
`#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208`

In the platformio.ini file I tried comparing the files from [this](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/firmware/Marlin-bugfix-2.0.x/platformio.ini) bigtreetech branch with the vanilla marlin branch and found a few differences:

Bigtreetech [common]:
`U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
TMCStepper@<1.0.0
Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip`

vs Marlin [common]:
`U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
TMCStepper@>=0.5.0,<1.0.0
Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/master.zip`

As well as:
Bigtreetech [LPC1768]
`platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip
build_flags       = -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin/src/HAL/HAL_LPC1768/include -IMarlin/src/HAL/HAL_LPC1768/u8g ${common.build_flags}
U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
TMCStepper@<1.0.0`

Where as in Marlin [LPC1768]
`platform = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.1.zip
build_flags       = -DU8G_HAL_LINKS -IMarlin/src/HAL/HAL_LPC1768/include -IMarlin/src/HAL/HAL_LPC1768/u8g ${common.build_flags}
extra_scripts     = Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py
U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
TMCStepper=https://github.com/p3p/TMCStepper/archive/pr_lpctimingfix.zip`

I was not able to find a composition of all of those libraries to get them to compile for a working firmware with UART mode. Maybe someone else has any idea on that? 
I tried to replace the `TMCStepper = https://github.com/bigtreetech/TMCStepper` which did not work and also tried changing other archives which always resulted in an error with the HAL at compiling.

sthgardner, 2020-06-03:
Used latest build

closed 2020-06-03 by sthgardner