bigtreetech/BIGTREETECH-SKR-V1.3 #127: MINI UPS IS USELESS

stefi01, 2019-11-13:
i wired up the mini ups to one of  the end stop max pin's on the skr 1.3 as shown but when there is power loss it ignores it and uses the stored power to try keep printing and then stops with out lifting z or doing anything and still will melt the print, am i missing something?  
also tried using p2_00 servo pin, same result, nothing, if i change high to low then it triggers a loss and halts the printer so i know it is wired up correctly 

define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define POWER_LOSS_PIN         P1_28 // Pin to detect power loss
    #define POWER_LOSS_STATE     HIGH // State of pin indicating power loss
    #define POWER_LOSS_PULL           // Set pullup / pulldown as appropriate
    //#define POWER_LOSS_PURGE_LEN   20 // (mm) Length of filament to purge on resume
    #define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.

    // Without a POWER_LOSS_PIN the following option helps reduce wear on the SD card,
    // especially with "vase mode" printing. Set too high and vases cannot be continued.
    #define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data
  #endif

boubouexpress, 2019-12-06:
M413 S1 in the boot code to enable

carl1961, 2019-12-11:
@boubouexpress boubouexpress can you share what/where to put in boot code, thanks

stefi01, 2019-12-12:
the latest version of marlin has an added few lines in config.adv which now does make this thing work and it apears on the lcd menu to turn it on or off, if you download the latest stable or bugfix or marlin 2.0 and scroll down to the lines i posted at the top of this page you see the extra lines which now allows this to work

milangupta1, 2020-11-01:
The latest Marlin-bugfix appears to manifest this behavior again. On start of print, the printer states "Power Loss - Halt - Please reset" if  PLR_ENABLED_DEFAULT is set to true. 
There are some conflicting directions on which pins to connect the UPS to for the skr 1.4 boards - P1_0 or P1_25. The pins_BTT_SKR_V1_4.h appear to define PS_ON_PIN as P1_0 and POWER_LOSS_PIN as also P1_0 (if not defined in configuration_adv.h).  Have tried connecting the UPS to both ports (E1_DET and PWR_DET), but both manifest this issue. 

How is this supposed to work ? I haven't been successful using the latest marlin bugfix builds ..

slmalb, 2020-11-16:
I am having the same problems with version 2.0.7.2 including bugfix version 2.0.x. Whatever I did, I could not succeed. On the other hand, the PW-DET (PC12) port on the skr mini e3 v2.0 motherboard is not defined in the pins_BTT_SKR_MINI_E3_V2_0.h file. Although I tried with different pins, the result did not change. At best, the unfinished print comes to its last position and does not start even though the screen says "printing".

javicarrera, 2021-01-24:
> The latest Marlin-bugfix appears to manifest this behavior again. On start of print, the printer states "Power Loss - Halt - Please reset" if PLR_ENABLED_DEFAULT is set to true.
> There are some [...]

Same issue, please anyone knows?

tiagomaricate, 2022-02-25:
Any news?
I'm trying to connect the relay and the ups module.
Could you people help me, please?
I don't have the STOPS pins or the E0/E1 pins free for use.
I'd like to understand how PS_ON and PWRDET pins work.
SKR 1.4 Turbo don't have the PS_ON port, just the PWRDET.