bigtreetech/BIGTREETECH-SKR-V1.3 #129: The first screen of Marlin 2.0 is showing some horizontal lines, appearing to be broken

dvdsouza, 2019-11-15:
We are using the branch bugfix-2.0.x and the first screen of Marlin is showing some horizontal lines, appearing to be broken.

[image]

After 3 seconds, appears the boot screen and everything works normally.

[image]

We have checked the power supply.
We have wrapped the "flat cable" in aluminum foil and ground everything correctly.
We have deleted the "firmware.bin" and it did not solve the problem.

The first time the firmware is loading the "firmware.bin", those horizontal lines do not appear, but when it is loading the "FIRMWARE.CUR" it does appear on each power-up.

ZIP file containing the Configuration.h and Configuration_adv.h files.
[marlin_config_files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3851647/marlin_config_files.zip)

bferrarese, 2019-11-15:
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "
> "name": "View Issue"
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",

That is normal. Just video memory before it is initialized (zeroed).

dvdsouza, 2019-11-19:
Well, this issue only appears in the bugfix-2.0.x branch, the 1.1.x branch works fine.
Do you have an idea of the root cause?