bigtreetech/BIGTREETECH-SKR-V1.3 #130: Error: 'LPC176x' has not been declared

dvdsouza, 2019-11-15:
We are trying to compile the firmware, but we are receiving the following errors:

[image]
[image]
[image]

Does Anyone Have a Clue about this problem?

Redssnake77, 2019-11-15:
Have you set your board type in the platform.io file? it towards the bottom, beneath your marlin files. Have you watched TeachingTech's YouTube video?

ursoft, 2019-11-16:
Cause: https://github.com/teemuatlut/TMCStepper/commit/a9187feee034c97ec292cf916d3a12d100774b0d

ursoft, 2019-11-16:
To avoid cause, you can use lower `TMCStepper@>=0.5.2,<0.6.0` as in original Marlin

BugfixXxX, 2019-11-18:
Hi guys, I am getting the same errors, I think it occures after I updatet the TMCStepper library but I also tried to lower the version which sadly doesn´t have any affect.
Does anyone have another solution?

dvdsouza, 2019-11-18:
@ursoft 

We have tried to change the version:
`TMCStepper@>=0.5.2,<0.6.0`
also the current bugfix 2.0 version:
`TMCStepper@>=0.5.0,<1.0.0` 

It makes sense, but we get the same error.
Another idea?

ursoft, 2019-11-19:
> @ursoft
> 
> We have tried to change the version:
> `TMCStepper@>=0.5.2,<0.6.0`
you should also delete ...\firmware\Marlin-bugfix-2.0.x\.pio\libdeps\LPC1768\TMCStepper* folder

Al9900, 2019-11-19:
Hi guys, i have your same problem but i've just solved... TMC stepper V.6.0.1 has been released 1 days ago and V.0.6.1, 6 days ago and both two create errors... Every time you compile the firmware , it'll always download the latest version and if you unistall TMC_stepper in platoformio libraries , Marlin will provide to reinstall it while compiling... The only way to get out is to download TMC_stepper 0.5 and put manually this  librarie inside Platoformio folder in USERS. 

Now i got 0 errors and it work to me, hope it'll help you.
 Fell free to contact me if was not clear. 

Cheers.

dvdsouza, 2019-11-19:
@Al9900 
It worked!
The problem is the version 0.6.x.
We have changed the platformio.ini with the version 0.5.2 `TMCStepper@>=0.5.2,<0.5.3`, and it worked like a charm.

I am closing the issue. Thanks.

dvdsouza, 2019-11-19:
The problem is the TMCStepper version 0.6.x.
We have changed the platformio.ini with the version 0.5.2 TMCStepper@>=0.5.2,<0.5.3

closed 2019-11-19 by dvdsouza

redeyejoe, 2020-01-02:
I installed .5.2 and changed the lib_dep to the TMCStepper@>=0.5.2,<0.5.3 in platform.ini  saved, but when compiling it still says it using |-- <TMCStepper> 0.6.1 and wont compile.

dolfelt, 2020-01-03:
@redeyejoe Make sure to update the dependency in the `[env:LPC1768]` section (further down the file)

KuraiKitsune, 2020-01-03:
> @redeyejoe Make sure to update the dependency in the `[env:LPC1768]` section (further down the file)

That was my missing part. All good now!
Many thanks, contributors!

redeyejoe, 2020-01-03:
Yes, thanks. That did it

suckmo, 2020-03-06:
[image]
[image]
[image]
[image]
Alright guys, Ive changed the ini file to call TMC 0.5.2 as you can see from the attachments. And it confirms in the lib dependecy that its calling the correct version, but its still failing.... any other suggestions???

enry68, 2020-03-21:
> Hi guys, i have your same problem but i've just solved... TMC stepper V.6.0.1 has been released 1 days ago and V.0.6.1, 6 days ago and both two create errors... Every time you compile the firmware , [...]

Thank you very much!!! I've solved installing manually the TMClibrary V.5.0. Thank you

ltmhenrique, 2020-04-09:
> Olá pessoal, tenho o mesmo problema, mas acabei de resolver ... O TMC stepper V.6.0.1 foi lançado há 1 dia e V.0.6.1, 6 dias atrás e os dois criam erros ... Sempre você compila o firmware, ele sempre baixa a versão mais recente e se você desinstalar o TMC_stepper nas bibliotecas platoformio, o Marlin fornecerá a reinstalação durante a compilação ... A única maneira de sair é fazer o download do TMC_stepper 0.5 e colocar manualmente essa biblioteca Pasta Platoformio em USUÁRIOS.
> 
> Agora eu tenho 0 erros e funciona para mim, espero que ajude você.
> Não hesite em contactar-me se não estiver claro.
> 
> Felicidades.

you could show more step by step how to do this, i'm a layman i'm struggling to solve this problem.

> Olá pessoal, tenho o mesmo problema, mas acabei de resolver ... O TMC stepper V.6.0.1 foi lançado há 1 dia e V.0.6.1, 6 dias atrás e os dois criam erros ... Sempre você compila o firmware, ele sempre baixa a versão mais recente e se você desinstalar o TMC_stepper nas bibliotecas platoformio, o Marlin fornecerá a reinstalação durante a compilação ... A única maneira de sair é fazer o download do TMC_stepper 0.5 e colocar manualmente essa biblioteca Pasta Platoformio em USUÁRIOS.
> 
> Agora eu tenho 0 erros e funciona para mim, espero que ajude você.
> Não hesite em contactar-me se não estiver claro.
> 
> Felicidades.

you could show more step by step how to do this, i'm a layman i'm struggling to solve this problem?

ltmhenrique, 2020-04-09:
> Hi guys, i have your same problem but i've just solved... TMC stepper V.6.0.1 has been released 1 days ago and V.0.6.1, 6 days ago and both two create errors... Every time you compile the firmware , [...]

you could show more step by step how to do this, i'm a layman i'm struggling to solve this problem?

Ozista, 2020-05-18:
Hi, Guys, I got this solved today without being able to download, but TMC_stepper 0.5.2 was installed in "Platform-Libraries-TMCstepper by tee..." anyway. 
I have replaced 4 occurrences of TMCStepper@>=.... to  "TMCStepper@>=0.5.2,<0.6.0"  in Platformio.ini.
This was my breakthrough.

Thank you all !!!