bigtreetech/BIGTREETECH-SKR-V1.3 #131: Connect TFT 3.5 v2 to SKR 1.3

joydi, 2019-11-15:
Hi everybody,

can someone help me i can manage it to connect the bigtreetech tft3.5 v2 to skr 1.3 mainbord.

do i have to set something in marlin or in the screen ? or both ?

Thanks in advance

Redssnake77, 2019-11-15:
to use the LCD mode you need to set it to ReRap Full graphics display in Marlin. To use the touch screen you just need to connect the 5 wire lead.

joydi, 2019-11-15:
Hi thank you for the answear but this also dont work if i just connect the 5 wire hell tells me no connection to the printer and when i i go in marlin and set #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

it comes the same error.

joydi, 2019-11-15:
OK i have solved it 

in marlin configuration.h 
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

and both devices should have same bautrate

Redssnake77, 2019-11-15:
that's what I have.