bigtreetech/BIGTREETECH-SKR-V1.3 #133: missing binary operator before token "("

Djinn1, 2019-11-16:
Any knows why I getting lots of this errors when trying to save configuration.h and .adv?

[image]

bferrarese, 2019-11-16:
I’m guessing you have typo in a #if statement prior to your first error occurrence.

Djinn1, 2019-11-16:
I am using Marlin bugfix-2.0.x. stock.

Djinn1, 2019-11-16:
> 
> 
> I’m guessing you have typo in a #if statement prior to your first error occurrence.

I am using Marlin bugfix-2.0.x. stock.

Bolivyne, 2019-11-17:
I am too getting these errors. New start up for my Ender 5, unable to get past this step of firmware update.

kepbrown, 2019-11-17:
Make sure that your Atom build doesn't have any issues. I was receiving those same errors and the issue turned out to be with the Atom upgrade caused some PlatformIO modules to not load since they were built for a prior version of Atom. For me it showed up as a problem on the bottom menu bar and walked me through rebuilding it for the installed Atom. I believe it also showed up in the Incompatible Packages tab.

Djinn1, 2019-11-17:
I am getting this errors when trying to save and build. On button is where the errors appears. I don't understand how I can check if something is wrong with the packages?