bigtreetech/BIGTREETECH-SKR-V1.3 #136: Question: Possible to flash over USB connevtion

TheLion, 2019-11-19:
Hi,

Just got the SKR 1.3 working in my Sidewinder X1. Works great!

But the Sidewinder doesn't has a slot for the SD card on the SKR. It doesn't even fit in the case. with the SD card in the slot. 

Taking a Dremel to it would be a solution, but is there a possibility to flash firmware to the board using the USB connection, when the SD card is not inserted?

karabas2011, 2019-11-19:
I use USB for flashing. No problem. Copy firmware to printer's storage and reboot

TheLion, 2019-11-19:
> 
> 
> I use USB for flashing. No problem. Copy firmware to printer's storage and reboot

And by printer's storage, you mean the SD card?

karabas2011, 2019-11-19:
Yes, microsd  automounting with printer connects via USB