bigtreetech/BIGTREETECH-SKR-V1.3 #137: Z endstopp does not trigger

joydi, 2019-11-21:
Hi i have my new skr 1.3 and have a problem. 

The minimal Z endstopp does not trigger.

I use marlin 2 

i followed this tutorial https://www.youtube.com/watch?v=EqpBEGJ5O_U&t=238s .

But the Z endstopps always has the same state.

Somebody has a solution ?

Thank you

rainerschulte, 2019-11-24:
In my case it was a matter of trial and error...

/**
 * Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *
 * Enable this option for a probe connected to the Z Min endstop pin.
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

-------------------
// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.

I'm using the Hardware from Stock Anet A8, so the "Connectors" have two pins only. PLUS is the right Pin, MINUS the middle, and the left Pin is kept untouched.

To test the connection I drove the X and Y Axis to the Middle of my Heatbed and Z far away up.
Using Pronterface I entered M119 and got all Endstops "OPEN". Then I pushed the Endstops manually by Hand and repeated the M119. The Endstop I pushed was now triggered... Fine so far.
Then I homed each Axis after another and found it working. 

For what ever reason, today I crashed the Endstopholder from Z-Axis...

nloding, 2019-11-30:
@rainerschulte what do you mean "right" and "left" pins? When you have the board oriented with the stepper drivers at the top, is it the top or bottom pin on your endstop plugs that is empty? I'm having an issue with my endstops ...

rainerschulte, 2019-12-01:
Hi Nathan,

 

my Setup now is, when the Drivers are on top, that the top Pins are empty and the bottom Pins have the positive Cables (in my case red) and the middle Pins have black (-) wires.

It took a lot of practice to crimp the JST-Connectors so, that the Wires stay in the Connectors when plugged in... :-(

 

The same Issue with Polarity is with the Fans. Currently I have both Fans on the Extruder for Steppermotor and Nozzle Cooling active. Cooling for the Board is in progress.

 

There are some Videos on Youtube where I found out to wire the Endstops correct. Now with all connectors soldered and crimped correct, I was able to print the stock housing for the SKR 1.3 without Issues like "Communication Errors" caused by wobbling connectors. The first Print yesterday via USB direct from PC, the Top of  the Case is currently running from the SD-Card in the Display.

 

I have ordered a BIGTREE TFT35 because my ANET-Display didn`t work wich polarity I ever chose. I think, the WiFi Module will follow soon. Actually I'm monitoring the Printer via OctoPi and Webcam.

Next Project to Print is a Housing for the Display I found on Thingiverse.

 

I'm so glad and happy, that this thing is now doing what I expect...
 

-- 
Der Wahnsinn hat einen Namen!

Rainer (Wahnsinn) Schulte

 
 

Gesendet: Samstag, 30. November 2019 um 22:41 Uhr