bigtreetech/BIGTREETECH-SKR-V1.3 #140: SKR 1.3 pinout overlay

Anarasha, 2019-11-23:
[image]

fxh1357, 2019-11-25:
Descriptions for the pins below the stepper drivers would be nice^^I was just recently searching for that information :)

Anarasha, 2019-11-25:
> Descriptions for the pins below the stepper drivers would be nice^^I was just recently searching for that information :)

You mean the ones literally under the drivers, between the two rows you plug the driver into? I'm not sure if those even have pinout names as they are primarily used to decide the steps for your drivers as far as I understand. I at least didn't see any pinout names on the diagram, but I could be wrong. 
If anyone can clarify the pinout names of those I'll be happy to add it in

fxh1357, 2019-11-27:
The pin assignment should be as following:

|red|red|black|
|-|-|-|
|MOT_MOSI|MS0|VDD|
|MOT_SCK|MS1|VDD|
|?-RX (CS)|MS2|VDD|
|MOT_MISO|RST/SLP|VDD|

Where `?` is [`X`, `Y`, `Z`, `E0`, `E1`] and `VDD` ist `3.3V`

|header|LPC1768|
|-|-|
|E1-RX|P1.1/ENET_TXD1|
|E0-RX|P1.8/ENET_CRS|
|Z-RX|P1.10/ENET_RXD1|
|Y-RX|P1.15/ENET_REF_CLK|
|X-RX|P1.17/ENET_MDIO|
|MOT_MISO|P0.5/I2SRX_WS/TD2/CAP2.1|
|MOT_SCK|P0.4/I2SRX_CLK/RD2/CAP2.0|
|MOT_MOSI|P4.28/RX_MCLK/MAT2.0/TXD3|