bigtreetech/BIGTREETECH-SKR-V1.3 #142: Bed not heating and LCD "ANET Full Graphics" is black

rainerschulte, 2019-11-23:
Hi Folks,

I have SKR 1.3 with TMC2208 in UART connected to my ANET PK8.
Unfortunately the Display keeps blank/dark, and shows no reaction. Its the one from ANET A6.

furthermore the BED is not heating in Pronterface. 
I connected the BED-Wires accordingly +/- to the SKR. The Thermistor I have connected to it's socket using Patchwires from Arduino, because the Thermistorplug is 3pins, as well as the Endstops... I have connected them temporarily with these Patchwires.

So Homing is working, the Steppers are silent and run where they should.

The Hotend is wired accordingly and heats up.

Kind regards
Rainer

rainerschulte, 2019-11-24:
Update: Bed is now working.

fxh1357, 2019-11-27:
Out of curiosity - What've you done to fix your heated bed?^^

Regarding your display it would be nice to know what you've configured in Marlin, which cable did you use? Is there an SD card slot in that display? Maybe you've mixed EXP1 and EXP2?

rainerschulte, 2019-11-28:
Hi,

 

i have disabled PID for the Bed. With PID it got max 50°C, without it can reach 65 and more. The PID-Settings come from the ANET A8 Configuration.

Yesterday I tested PID Tuning, without success. The Temerature raises up to 35° during the first pass.

Because of time i aborted the Process.

 

Concerning the Display I ordered a TFT35 Controller, wich is already installed and running. Printing from both SD-Cards and via USB seems working.

 

These days I was able to print a Socket for the board. As a result i found out, that this Print was the best I ever had.. until homing the X-Axis. The Motor was very hot and noisy.

Today I changed the Motor and found, that it wasn't the Motor. I flashed the CONFIGURATION.h and the Adv. one several Times. But the Axis ran too quick and too far. So I experimented with the Settings.

 

I must change the Computer to attach the Files.

 

Currently the Printer is running, but the X-Motor is very hot again.

 

It would be great, if you have an Idea....
 

-- 
Der Wahnsinn hat einen Namen!

Rainer (Wahnsinn) Schulte

 
 

Gesendet: Donnerstag, 28. November 2019 um 00:48 Uhr

closed 2020-02-22 by rainerschulte