bigtreetech/BIGTREETECH-SKR-V1.3 #144: tft24-v1.1

Connor445, 2019-11-23:
i just got a tft screen and went to use it with my printer after i installed all the firmwear and my marlin would not compile it said the pins are not declared in this scope

In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:42:
Marlin\src\lcd\dogm\ultralcd_DOGM.h:178:21: error: 'FSMC_CS_PIN' was not declared in this scope
   #define U8G_PARAM FSMC_CS_PIN, FSMC_RS_PIN
                     ^~~~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'

 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:178:21: note: suggested alternative: 'E1_CS_PIN'
   #define U8G_PARAM FSMC_CS_PIN, FSMC_RS_PIN
                     ^~~~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:178:34: error: 'FSMC_RS_PIN' was not declared in this scope
   #define U8G_PARAM FSMC_CS_PIN, FSMC_RS_PIN
                                  ^~~~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:178:34: note: suggested alternative: 'PS_ON_PIN'
   #define U8G_PARAM FSMC_CS_PIN, FSMC_RS_PIN
                                  ^~~~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
 U8G_CLASS u8g(U8G_PARAM);
               ^~~~~~~~~
*** [.pio\build\LPC1768\src\src\lcd\dogm\ultralcd_DOGM.cpp.o] Error 1
========================= [FAILED] Took 36.22 seconds =========================

dose any one know what i chould do? thanks

3ddev-j, 2020-01-05:
I've got the same problem

Connor445, 2020-01-05:
No solution I’m not really looking my self I’m pretty busy but if you or anyone fine a solution PLEASE SHARE. It would be greatly appreciated.

Connor445, 2020-01-05:
Find*

3ddev-j, 2020-01-05:
I just realized the solution, instead of looking for the TFT24 on Marlin you'll need to upload a separete firmware just for the TFT24 as it has it's own processor.
What i did was:

In marlin i have selected FULL GRAPHICS SMART CONTROLLER, you can choose none grafical controller but if you put this you'll not be able to switch LCD modes (TFT or LCD).

Go to this Git : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
Download all files, in folder " Copy to SD Card root.." copy to an SD card only BIGTREE_TFT24_V1.1.25.bin file, TFT24 and Firmware archive folders.

Plug your SD Card on your TFT24, power it and wait for upload.

If you choose to switch lcd mode, you'll will need both LCD and TFT cable connected to your SKR v1.3
Just hold enconder button and it will appear .

Connor445, 2020-05-17:
OK i did that and the firmware on the LCD is not a problem its all configured correctly the problem is in the configuration in the skr's firmware and the firmware dose not compile and gives me the error message above.

0xD34DC0DE, 2020-06-03:
I have the same problem too