bigtreetech/BIGTREETECH-SKR-V1.3 #145: TMC2209 drivers not communicating

LordBBQ, 2019-11-24:
I have recently upgraded to a SKR 1.3 with 5 BIGTREETECH TMC2209 drivers. The firmware is all configured, but the LCD shows 'TMC Connection error'. M122 also shows this. 
[image]

fxh1357, 2019-11-25:
Your pin configuration looks like you have set E0 and E1 as well in UART mode using the jumpers below the stepper ports - is this reflected in your firmware configuration as well? No driver in those ports could be interpreted as "TMC connection error".

If that's not the case some detailed information would be necessary - your changes to the firmware, the firmware version, your driver version, pin configuration and M122 printout, etc.

Ru83nG, 2020-01-07:
I had been having that issue showing up on screen.
I used my bench test power supply and fed it 12VDC instead of the 5VDC from the usb cord.
That fixed my problem.  Of course, I had the drivers installed.
Remember to change the jumper tho.