bigtreetech/BIGTREETECH-SKR-V1.3 #146: Z_Min always triggered (

fineobjetcs, 2019-11-27:
Hi ! I have this issue
At start all be ok! Z_Min works as Z_Probe fine, but three days ago Z_Min starts always triggered. I try to change firmware (Marlin20bugfix, smoothiware), change settings(invert ZMIN Probe etc) but nothing happens, i try with and without wires and sensor.
What wrong ? May it be electrical error ? How i check the PIN directly ? Maybe from arduino IDE ? With simple code ? PIN 1_25 is analog or digital ?

fxh1357, 2019-11-28:
"Three days ago" sounds like a limit switch connected as normally closed, thus a broken wire or disconnected switch would result in a 'TRIGGERED' state

oldman4U, 2020-08-27:
Close this ticket or reply.

closed 2020-09-02 by fineobjetcs