bigtreetech/BIGTREETECH-SKR-V1.3 #147: X- and Y- Endstop stuck at OPEN or TRIGGERED depending on inversion settings

bendavepaul, 2019-11-27:
I am having trouble with my SKR v1.3.
X- and Y- are either always triggered or open depending on how I set the inversion in Marlin and they do not react to the endstop being triggered. I tried different kinds of endstops and they are all stuck at triggered (not because I set the logic wrong in marlin). Only Z- is working fine.

Without any endstops plugged in and running M119 on Pronterface I get:
y_min OPEN
x_min OPEN
z_min TRIGGERED 

I then used a female to female Dupont Wire to connect the S pin to the G pin On X- and sent another M119 to Pronterface, I get the same:
y_min OPEN
x_min OPEN
z_min TRIGGERED 

I then connected the S pin to the V pin and sent another M119 and got the same:
y_min OPEN
x_min OPEN
z_min TRIGGERED

I then repeated with Y- and get the same results. Note the Z- is working as normal.

Do you think it's a firmware problem or with the board?
I've attached my config.h file.

[Configuration.h.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3895419/Configuration.h.zip)

fxh1357, 2019-11-27:
Have you changed something to the ENDSTOPPULLUPS in configuration.h? How are your endstops wired in normally-closed or normally-opened configuration?

bendavepaul, 2019-11-27:
Nope, the ENDSTOPPULLUPS are fully commented out.  Z endstop is wired to normally open and X and Y endstops seem to be wired to normally closed, steppers would only move tiny amounts until I inverted the signal interpretation.
#define X_MIN_ENDSTOP_INVERTING true 
#define Y_MIN_ENDSTOP_INVERTING true
#define Z_MIN_ENDSTOP_INVERTING false

Now they move without a problem, but don't stop.

fxh1357, 2019-11-27:
Could you try to uncomment ENDSTOPPULLDOWN_XMIN and run a M119 without any connections made? The outcome should be TRIGGERED when X_MIN_ENDSTOP_INVERTING true

bendavepaul, 2019-11-28:
I followed this and uncommented ENDSTOPPULLDOWN_XMIN, ran a M119 with nothing connected and got OPEN.

fxh1357, 2019-11-28:
So regardless of ENDSTOPPULLDOWN_XMIN or ENDSTOPPULLUP_XMIN you always get OPEN? (With no change to the inverting or that pin)

bendavepaul, 2019-11-28:
Yes that's correct, no change between ENDSTOPPULLDOWN_XMIN or ENDSTOPPULLUP_XMIN. It's always reading OPEN.

fxh1357, 2019-11-28:
Have you changed something to the pin assignments? The board is correctly set to BOARD_BIGTREE_SKR_V1_3? X_HOME_DIR is set to -1? X_MIN_PIN is P1_29 in pins_BTT_SKR_V1_3.h?

bendavepaul, 2019-11-28:
I've just double checked everything again including pins_BTT_SKR_V1_3.h and nothing has been changed to pin assignments, X_MIN_PIN is set to P1_29, the board is set to BOARD_BIGTREE_SKR_V1_3 and X_HOME_DIR is set to -1.

nloding, 2019-11-30:
@fxh1357 @bendavepaul I'm having the _exact same issue_ with my brand new SKR 1.3 board. I have nothing to add - I've confirmed continuity at the JST connector, so I know it's not an issue with the switch or the wires. Yet it still fails to trigger. I also found that the fans don't turn on, so I suspect a faulty board?

nloding, 2019-12-01:
@fxh1357 @bendavepaul I was able to resolve the endstop issue! It took a bunch of scouring and I eventually came across this issue/comment: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/3#issuecomment-486159992

The DIAG jumpers are grabbing the signal pin. I removed the first three jumpers (X, Y, Z) and it works now! I still have an issue with my fans not running, but the endstops work now!

bendavepaul, 2019-12-01:
Hi @nloding thanks for the suggestion. I tried removing the DIAG jumpers on (X,Y,Z) and sadly it hasn't solved the problem for me.  The bed and printer head are still crashing into the endstops.

661020786, 2021-08-18:
Interestingly, I'm facing the same problem but with the Reality v4.4.2 board on and Ender 5 Pro.  M119 reports X_MAX_ENDSTOP is triggered all the time (even when switch is manually closed) but when I invert it (X_MAX_ENDSTOP_INVERTING -> true), the printhead moves in the correct direction during auto-homing, but keeps crashing into the frame.    

I haven't found a fix yet -- maybe I'll replace the endstop switch with another from another printer

661020786, 2021-08-18:
So, I just tried it with a different switch (I used the now-unused z-stop switch) and no change.   In the normal configuration (X_MAX_ENDSTOP_INVERTING false) it reports TRIGGERED whether the switch is open or closed.  It's either a wiring issue or, in my Creality v4.2.2. case, a motherboard issue.

nloding, 2021-08-19:
@661020786 are you using a BTT SKR v1.3 board? Or are you using the Creality v4.2.2 board?

TrollDas, 2024-09-04:
I came here facing the same issue but with SKR3 board; im using normal swithces in NC mode, for X_MIN and Y_MAX, (modded ender5+), the issue doesnt appear to eb resolved..
I did end up once on the official marlin page for endstop config where it states : 

{Regardless of the type the basic way of working is the same:

A typically 5 Volt signal (HIGH) drops to 0 Volt (LOW): Normally closed (NC) switch
A 0 Volt signal (LOW) rises to 5 Volts (HIGH): Normally open (NO) switch }

My guess, for the 2 wires coming from the switch needs to go to Vcc(5v) and Signal on the board, instead of Signal and ground..

NOW, i dont have the guts to actually try in fear of frying the board , but tell me if iam wrong?