bigtreetech/BIGTREETECH-SKR-V1.3 #148: SKR V1.3 - TMC2208 dual z stepper problem.

bogachansengun, 2019-11-27:
I am trying to configure marlin bugfix 2.0.0 ( both from bigtreetech github repo an marlin github repo ). 

If I configure like this ; 

#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2208
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2208
//#define E1_DRIVER_TYPE TMC2208
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

I have TMC Connection Error for Z2. If I configure like this ;

#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE TMC2208
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2208
#define E1_DRIVER_TYPE TMC2208
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988

there is no TMC Connection Error and Z axis motors work but not syncnorised.  I tried to adjust with 
#define DISTINCT_E_FACTORS
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 93, 400 }

But in this case I got compile error , too many parameters for DEFAULT_AXIS_STEPS_PER_UNIT

Is there a solution for this problem ? or is this a bug for TMC2208 ?

bogachansengun, 2019-11-27:
M122 output for two cases are below ;

#define Z2_DRIVER_TYPE TMC2208 ;

12:14:23.531 : X	Y	Z	Z2	E
12:14:23.608 : Enabled		false	false	false	false	false
12:14:23.608 : Set current	580	580	580	800	650
12:14:23.664 : RMS current	550	550	550	1436	642
12:14:23.719 : MAX current	776	776	776	2025	905
12:14:23.719 : Run current	17/31	17/31	17/31	25/31	20/31
12:14:23.719 : Hold current	8/31	8/31	8/31	12/31	10/31
12:14:23.777 : CS actual	8/31	8/31	8/31	0/31	10/31
12:14:23.833 : PWM scale	10	10	10	0	12
12:14:23.890 : vsense		1=.18	1=.18	1=.18	0=.325	1=.18
12:14:23.945 : stealthChop	true	true	true	false	true
12:14:24.002 : msteps		16	16	16	256	16
12:14:24.058 : tstep		max	max	max	0	max
12:14:24.058 : pwm
12:14:24.058 : threshold
12:14:24.058 : [mm/s]
12:14:24.113 : OT prewarn	false	false	false	false	false
12:14:24.169 : off time	4	4	4	0	4
12:14:24.225 : blank time	24	24	24	16	24
12:14:24.226 : hysteresis
12:14:24.282 : -end		2	2	2	-3	2
12:14:24.337 : -start		1	1	1	1	1
12:14:24.337 : Stallguard thrs
12:14:24.337 : DRVSTATUS	X	Y	Z	Z2	E
12:14:24.394 : stst		*	*	*		*
12:14:24.450 : olb
12:14:24.506 : ola
12:14:24.562 : s2gb
12:14:24.618 : s2ga
12:14:24.675 : otpw
12:14:24.730 : ot
12:14:24.787 : 157C
12:14:24.842 : 150C
12:14:24.898 : 143C
12:14:24.954 : 120C
12:14:24.954 : s2vsa
12:14:24.954 : s2vsb
12:14:24.954 : Driver registers:
12:14:24.963 : X	0xC0:08:00:00
12:14:24.972 : Y	0xC0:08:00:00
12:14:24.981 : Z	0xC0:08:00:00
12:14:25.002 : Z2	0x00:00:00:00	 Bad response!
12:14:25.010 : E	0xC0:0A:00:00
12:14:25.020 : Testing X connection... OK
12:14:25.029 : Testing Y connection... OK
12:14:25.038 : Testing Z connection... OK
12:14:25.060 : Testing Z2 connection... Error: All LOW
12:14:25.067 : Testing E connection... OK

#define E1_DRIVER_TYPE TMC2208 ;

12:18:50.254 : X	Y	Z	E
12:18:50.290 : Enabled		false	false	false	false
12:18:50.290 : Set current	580	580	580	650
12:18:50.326 : RMS current	550	550	550	642
12:18:50.362 : MAX current	776	776	776	905
12:18:50.362 : Run current	17/31	17/31	17/31	20/31
12:18:50.362 : Hold current	8/31	8/31	8/31	10/31
12:18:50.398 : CS actual	8/31	8/31	8/31	10/31
12:18:50.434 : PWM scale	10	10	10	12
12:18:50.470 : vsense		1=.18	1=.18	1=.18	1=.18
12:18:50.506 : stealthChop	true	true	true	true
12:18:50.541 : msteps		16	16	16	16
12:18:50.577 : tstep		max	max	max	max
12:18:50.577 : pwm
12:18:50.578 : threshold
12:18:50.578 : [mm/s]
12:18:50.613 : OT prewarn	false	false	false	false
12:18:50.649 : off time	4	4	4	4
12:18:50.684 : blank time	24	24	24	24
12:18:50.684 : hysteresis
12:18:50.720 : -end		2	2	2	2
12:18:50.756 : -start		1	1	1	1
12:18:50.756 : Stallguard thrs
12:18:50.756 : DRVSTATUS	X	Y	Z	E
12:18:50.791 : stst		*	*	*	*
12:18:50.827 : olb
12:18:50.863 : ola
12:18:50.898 : s2gb
12:18:50.934 : s2ga
12:18:50.969 : otpw
12:18:51.005 : ot
12:18:51.041 : 157C
12:18:51.076 : 150C
12:18:51.112 : 143C
12:18:51.148 : 120C
12:18:51.148 : s2vsa
12:18:51.148 : s2vsb
12:18:51.148 : Driver registers:
12:18:51.157 : X	0xC0:08:00:00
12:18:51.166 : Y	0xC0:08:00:00
12:18:51.175 : Z	0xC0:08:00:00
12:18:51.184 : E	0xC0:0A:00:00
12:18:51.193 : Testing X connection... OK
12:18:51.201 : Testing Y connection... OK
12:18:51.210 : Testing Z connection... OK
12:18:51.219 : Testing E connection... OK

fxh1357, 2019-11-27:
First of all, the Z2 definition should be correct - using Z2 as E1 sound hacky to me. Have you set Z_DUAL_STEPPER_DRIVERS in the advanced config file? Where does the current settings of 580 come from for x, y and z?

bogachansengun, 2019-11-27:
> First of all, the Z2 definition should be correct - using Z2 as E1 sound hacky to me. Have you set Z_DUAL_STEPPER_DRIVERS in the advanced config file? Where does the current settings of 580 come from for x, y and z?

Ofcourse I enabled Z_DUAL_STEPPER_DRIVERS  .  But does not work.  I dont know where 580 come from but I think its default for TMC2208.

fxh1357, 2019-11-27:
I guess you've used the Ender3 example config as a base? That would explain those current values of 580. The second printout (#define E1_DRIVER_TYPE TMC2208 ;) can't be correct - there is a missing entry for E1, are you sure that everything was uploaded to the board as expected?

bogachansengun, 2019-11-28:
Yes. You are right. I used Ender3 example file as base. I agree with you , second printout have missing info about E1 . I am sure all uploaded as expected. I think this is a bug of Marlin. I tried with TMC2100
today and result is same. :-(  I am thinking about returning back to MKS GEN V1.4 :(

fxh1357, 2019-11-28:
As far as I know TMC2100's have neither SPI not UART capabilities, so what exactly have you tested in this regard?
You have removed all debug jumpers (bottom left of the stepper drivers port array), set all UART Jumpers and removed every jumper below the stepper driver, right? Maybe try to switch your Z2 driver with X to make sure that it's not the driver