bigtreetech/BIGTREETECH-SKR-V1.3 #149: [Question] TMC2100 jumpers setup

huczas, 2019-11-27:
Hi, I have TMC2100 modules v1.3 from fysetc.com. And in [PDF](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BIGTREETECH%20SKRV1.3%20guide%202019%20-%206.pdf) I found a table at page 18. 
But what do I not understand is how to connect jumper to GND and second to V+ (open - is obvious just no jumper at all).
In TMC2100 there is used only MS1 and MS2(depends on producer CFG1 and CFG2). 

CFG1 | CFG2 | Schrittteilung | Interpolation | Mode
-- | -- | -- | -- | --
GND | GND | 1 (full step) | none | spreadCycle
VCC | GND | 2 (half-step) | none | spreadCycle
open | GND | 2 (half-step) | 256 µ-Schritte | spreadCycle
GND | VCC | 4 (quarter-step) | none | spreadCycle
VCC | VCC | 16 µ-steps | none | spreadCycle
open | VCC | 4 (quarter-step) | 256 µ-steps | spreadCycle
GND | open | 16 µ-steps | 256 µ-steps | spreadCycle
VCC | open | 4 (quarter-step) | 256 µ-steps | stealthChop
open | open | 16 µ-steps | 256 µ-steps | stealthChop

fxh1357, 2019-11-27:
Considerung https://wiki.fysetc.com/TMC2100/ I guess these modules are meant to be configured via soldering. Using the pin header below the driver board you'll only be able to make connections from CFG to HIGH; The left row of each 3x3 block is the SPI interface. Have a look below the driver boards before making any connections, if one of the soldered jumpers is a connection from - let's say CFG1 to ground and you connect MS0 to HIGH you would've basically shorted your board.

huczas, 2019-11-28:
Thank you, didn't know about that wiki, seems that there is everything what I need 🙂

closed 2019-11-28 by huczas