bigtreetech/BIGTREETECH-SKR-V1.3 #151: use tx & rx from exp1 or exp2 port to hook a raspberry pi via serial

ipatch, 2019-11-30:
I have a skr v1.3 and an raspberry pi i'd like to possibly hook up via the gpio on the raspberry pi and via the tx/rx on one of the exp1 or exp2 ports on the skr v1.3 board.  This would allow me to not have to use the USB port for connecting the pi, and also free up some much needed room within the case of my printer.

I have the btt tft35v2 hooked up to the tft port of the printer, so i cant use that dedicated pin header to wire up the pi to the board.

[image]

ipatch, 2019-12-10:
I was able to wire up the rpi to the btt tft35v2 so no need for wiring directly to the board, but on another note the **EXP1** and **EXP2** ports use _SPI_ and not _UART_ so that is a **total** NO GO.

closed 2019-12-10 by ipatch

nickcooley, 2020-02-08:
@ipatch  -- Just out of curiosity, what was your solution while using the btt tft35?

ipatch, 2020-02-08:
i've been jotting down notes below

https://github.com/ipatch/Marlin/tree/ibuild-marlin2-bugfix/media

and i believe have youtube videos explaining how i wired my pi to the btt tft35v2 to connect the screen and the pi to the board at the same time.