bigtreetech/BIGTREETECH-SKR-V1.3 #153: Board died

andrebianchessi, 2019-12-02:
I was configuring my printer and everything was going well. The hotend was heating up and the motors were moving correctly. All thermal protections were enabled on the configuration.h.

At one point, I increased the temperature at the hotend and, while the temperature was rising, set the motors to "auto-home".
As soon as I clicked to auto-home, the board "died". LCD, lights at print head, fans, stepper motors and everything else turned off. Now the board is unresponsive, and not even the red light turns on, even after I disconnected everything from the board and power it via USB (with the jumper in correct position). Also, the board won't install a new firmware if I transfer it to the SD card directly.

The last thing I noticed is that the processor and the AMS1117 (both marked in red on the picture I attached) get very hot when the board is connected using USB.

I am pretty sure the board is dead for good, but I wanted to know what might have cause this so that I can stop it from happening again.

[image]
.

yfaykya, 2019-12-04:
I am on board number 4 at moment. All died randomly :-(

ellensp, 2019-12-13:
Sounds like 3.3v regulator is dead and sent 5v to the 3.3v cpu...   very sad.   time for SKR 1.4.  These 32 bit boards are more sensitive to static damage than slower 8 bit boards...

tyty81601, 2019-12-13:
Are you even reading what i am saying? I will be posting your replies online and let thousands of people know how you respond too your customers. I asked “Hello i just seen that you guys have released a new motherboard and i was wondering if you could send me a stl or a step file of the new motherboard im having issues with github and i cannot download the file it downloads the step file as a text file and not a 3d model” i want a 3d file of the motherboard and your answering my question with something that’s totally off topic of what i asked

Pixel-Soup, 2020-01-30:
Plugging in a driver board incorrectly is the easiest way to kill the LPC chip.  My board died as well.  The MCU is drawing a huge current.  Don't know if any other components are damaged, but I will find out when I replace the LPC1768.

Pixel-Soup, 2020-02-01:
Replacing the lpc1768 fixed all problems on my board.

V3N0MX92, 2020-04-09:
> Replacing the lpc1768 fixed all problems on my board.

Did you have to upload the bootloader to the new LPC1768 ? If yes how did you do it.
Thanks in advance!

Pixel-Soup, 2020-04-09:
The ams1117 is probably not defective.  The 1768 is drawing lots of current and causing it to heat up.  The 1768 is somewhat difficult to replace without the right equipment.
I loaded the bootloader using a J-link on the ICP port.  The ST-link is a $5.00 alternative.  There are Chinese clones of the j-link, but they might not work except with old software.