bigtreetech/BIGTREETECH-SKR-V1.3 #154: No UART connection  with BIGTREETECH TMC2208 V3

thomasdr1, 2019-12-04:
Hi,
I get under Marlin 2 no UART connection to the TMC2208 drivers.
The jumpers on the board are set correctly.
The TMC2208 version 3 should not require solder bridges either.
Various settings in the configuration.h and CONFIGURATION_ADV.H are set.

Can someone provide me a working configuration?

Thanks Thomas

fxh1357, 2019-12-04:
Jumpers set correctly implies the removal of the Diag-Jumpers?^^

thomasdr1, 2019-12-04:
which is the diag jumper? i use only the red jumper under the socket...

thomasdr1, 2019-12-04:
i mean the the UART jumper...

wolfiejp, 2019-12-04:
Check your stepper Library,  i had the same issue had to roll back the
stepper driver library to 5.03 i think it was once i did this it solved the
issue.
Also make sure your changing power from usb to mains power or else you will
get the error no matter what

Good Luck

thomasdr1, 2019-12-05:
sorry, no effect
attached my config

[Marlin-bugfix-2.0.x TCM2308.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3928645/Marlin-bugfix-2.0.x.TCM2308.zip)

TimoBre, 2019-12-07:
Du you have set only the UART jumpers. Between the stepper legs have to be removed? 

[image]

thomasdr1, 2019-12-07:
the Stepper legs are removed, i tried bt and marlin original version, which Stepper library Version works?