bigtreetech/BIGTREETECH-SKR-V1.3 #155: E0 it's not working.. A4988

oleolsenn, 2019-12-06:
Can't get my E0 to work
I have received 2 death 2130 drivers, so i am trying to run my SKR 1.3 bort like..
```
#define X_DRIVER_TYPE  TMC2130_STANDALONE
#define Y_DRIVER_TYPE  TMC2130_STANDALONE
#define Z_DRIVER_TYPE  A4988
#define E0_DRIVER_TYPE A4988
```
XYZ is working 
If i swap the Z cable with E cable extruder stepper motor runs fine.
I have also tried switching the drivers around but it's still the same socket that is not working E0. Drivers and cabels is fine.

Please help me

Ph0rkoz, 2019-12-10:
Maybe this: https://github.com/Arksine/LPC17xx-DFU-Bootloader/releases/tag/0.4.1

mjaureguiar, 2020-01-12:
did you solve the problem? I have x y z tmc2208 and e0 a4988 and I can't make it work on SKR 1.4 Turbo.

jaburges, 2020-03-30:
Same here - TMC2209 on X,Y,Z and A4988 on E0, E1
Upgraded from MKS Gen v1.4

jaburges, 2020-03-30:
@bigtreetech can you take a look? Note on exactly the same firmware just changing:
X,Y & Z Stepper drivers from TMC2100 > TMC2209
Leaving E0 and E1
BOARD_BTT_v1_4_SKR_TURBO

Which was working very well on a RAMPS_GEN_14
The extruders lock up and don’t respond.

Jumpers for the A4988 are all on the right 2 columns. 
Please advise if the jumpers need changing or there is something specific in the firmware that needs changing?
(I also tested with a TMC2209 in E0 with the same jumper config as X,Y & Z and had the same lock up issue)

jaburges, 2020-03-31:
Ok, it seems enabling ‘MINIMUM_STEPPER_PULSE 1’
got it to work.
After seeking some guidance on the marlin discord you have to implicitly define the E0 and E1 drivers. 
Not defining pulse is 2, defined pulse is 1
(OP has this defined)

So that leaves step and micro step settings to check on the A4988.
The jumpers should be all 4 rows, across the middle and right columns for the A4988.
Good luck