bigtreetech/BIGTREETECH-SKR-V1.3 #157: Controller \ Firmware kills sd cards

thomasdr1, 2019-12-07:
I write the firmware directly from Visual Studio via USB to the controller.
This works 10 times. After that, copying the firmware is only possible via explorer. This is again a few times, then the SD card is broken. It is no longer formatted in any device.
I have already lost three SD cards (16, 32, 128GB). That costs more than the controller.
I use Windows 10 and the current Marlin 2 version.

In configuration.h: #define SDSUPPORT is set

Since the UART connection to the Bigtreetech TMC2208 V3 does not work, the controller has already cost me a lot of time.

TimoBre, 2019-12-07:
Bad Thing.

Do you activated the SD Card ONBARD ?
Please check the adv.h  (mine in Line 1153)
if this is set:
#define SDCARD_CONNECTION **ONBOARD**

uncomment and change it from LCD to ONBARD

I have no problems with BTT TMC2208 V3. What ist your problem?
Do you use the firmware fom BTT or an actual on from Marlin?

Hope it helps

billydragon, 2019-12-14:
My SKR V1.3 and TFT35 V3 have the same problem. It happen just today when i update lates Marlin Bugfix-2.0.x and TFT35 Firmware. SD card read only 1 time after formated and copy firmware.bin to card, put it to SKR v1.3 controller board then after controller updated new firmware the card lose format and can't access until reformat again

funbobby2001, 2020-10-15:
I just had this same issue with my V1.4 Turbo, which is less than one month old. Turned the printer on this morning and Marlin would not load on the LCD and the USB interface was dead. Tracked it down to the micro SD which would not show up when inserted directly into my desktop, even using methods specifically meant to access and repair SD cards that don't show in windows explorer; the card was completely fried.