bigtreetech/BIGTREETECH-SKR-V1.3 #160: TMC2130 All High

bluetangkid, 2019-12-08:
I just recently got the SKR V1.3 for the wiring simplicity and 32 bit stuff. Sadly, this has also created a few issues. The SPI jumpers are in the correct places and the +5V is in INT like it is supposed to be, but it doesn't work. If I swap through some different drivers sometimes there are all LOW, but most of them are HIGH. I am using the marlin version in this repo. Thanks for the help!
Here's the readout of M122
```
SENDING:M122
		X
Enabled		false
Set current	800
RMS current	795
MAX current	1121
Run current	25/31
Hold current	12/31
CS actual	31/31
PWM scale	255
vsense		1=.18
stealthChop	true
msteps		0
tstep		4294967295
pwm
threshold
[mm/s]
OT prewarn	true
OT prewarn has
been triggered	false
off time	15
blank time	54
hysteresis
-end		12
-start		8
Stallguard thrs	0
DRVSTATUS	X
stallguard		*
sg_result	1023
fsactive		*
stst		*
olb		*
ola		*
s2gb		*
s2ga		*
otpw		*
ot		*
Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
```

and with some drivers

```
SENDING:M122
		X
Enabled		false
Set current	800
RMS current	1436
MAX current	2025
Run current	25/31
Hold current	12/31
CS actual	0/31
PWM scale	0
vsense		0=.325
stealthChop	false
msteps		256
tstep		0
pwm
threshold
[mm/s]
OT prewarn	false
OT prewarn has
been triggered	false
off time	0
blank time	16
hysteresis
-end		-3
-start		1
Stallguard thrs	0
DRVSTATUS	X
stallguard
sg_result	0
fsactive
stst
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x00:00:00:00	 Bad response!
```

I only have X connected right now for testing purposes.

HilfsaffeMojo, 2019-12-09:
I have the same error, but motion and sensorless homing works fine.
What Version of TMC 2130 did you use?

bluetangkid, 2019-12-09:
Bigtree v3.0

HilfsaffeMojo, 2019-12-10:
I have solved the problem by cutting the not used DCO Pin of my TMC2130 V2 (Watterott)
[image]

ellensp, 2019-12-13:
The TMC2130 requires that the 12v/24v power be active from power on, not later than the 5v.

tyty81601, 2019-12-13:
What? That answer has nothing to do with the question i asked

Talha909, 2019-12-13:
> Bigtree v3.0

have you got any solution.  I have the same SKR 1.3 and TMV2130 V3.

Talha909, 2019-12-13:
> I have solved the problem by cutting the not used DCO Pin of my TMC2130 V2 (Watterott)
> [image]

can you share the config files

tyty81601, 2019-12-13:
ANSWER MY FUCKING QUESTION

Talha909, 2019-12-13:
you asked nothing,

MowMoney, 2019-12-13:
What is your problem? Maybe open your own issue and someone will answer you, rather than jumping into other issues and demanding answers to a question that nobody knows about.

bluetangkid, 2019-12-15:
Hey could you people chill out a little this is a github issues area not reddit. I connected 3 which now read HIGH, LOW, and HIGH. The pins are cut so I don't know what else to try

bluetangkid, 2019-12-15:
VIO is getting 3.3V and VM is getting 24V. Tried inverting stepper EN. The board or steppers must just be bad, and I'm not sure which one.

bluetangkid, 2019-12-15:
SDI, SCK, CS, and SD0 have no connection on 4 of my drivers but on one there is high resistance. I assume probably the 4 drivers are broken but the 1 high resistance one doesn't work either.

Dazzamatazz, 2019-12-28:
I have used  TMC2130 v2 with SKR v1.3 with no issues i did have to solder on the DIAG0 and DIAG1 pins

bluetangkid, 2019-12-31:
I bought 4 new drivers and none work, I'm assuming its a board problem but it's awful that I had to buy 3 sets of steppers and 3 boards to try to get it working.  @Dazzamatazz what do you mean by solder on the DIAG0 and DIAG1 pins?

Dazzamatazz, 2019-12-31:
My Big tree tech TMC2130 are version 1.1 and do not have pins soldered on DIAG0 and DIAG1.
I soldered pins on and have sensorless homing working.

bluetangkid, 2019-12-31:
Ah. Mine came already soldered

Dazzamatazz, 2019-12-31:
what firmware are you using?
I downloaded Marlin 2.0 from marlin and have edited it myself.

mcFactor, 2020-02-16:
Well, same here. I have SKR 1.3 and BegTree TMC2130 v2.0. no matter what, can't get sensorless homing working...

Dazzamatazz, 2020-02-16:
i have built a corexy and have got sensorless homing working but i have had a few false triggers on the y axis and nothing i have done has fixed it so i've now installed optical switches

Bruce17, 2020-04-28:
I have the same issue here as well. Can't get the TMC2130 to connect via SPI to my SKR v1.4 at all.

Marlin is build from scratch with only the bare minimum settings to get this board up and running. I had already the board installed in my printer and adjusted it to that as well. The steppers are working in some kind of fallback mode and they are quite loud. During operation the M122 also fails. I connected DIAG0 and DIAG1, but no luck as well. I did not cut the pin as described by @HilfsaffeMojo in https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/160#issuecomment-564213644. I also tried to power up the bare minimum setup using an external 12 V power supply, but no luck either.

Details:
Drivers: BigTreeTech TMC2130 v3.0
Board: BigTreeTech SKR v1.4 Turbo

<details>
<summary>M122 output (one driver installed only):</summary>
<p>
M122
		X
Enabled		false
Set current	800
RMS current	795
MAX current	1121
Run current	25/31
Hold current	12/31
CS actual	31/31
PWM scale	255
vsense		1=.18
stealthChop	true
msteps		0
tstep		4294967295
PWM thresh.	
[mm/s]		
OT prewarn	true
triggered
 OTP		false
off time	15
blank time	54
hysteresis
 -end		12
 -start		8
Stallguard thrs	0
DRVSTATUS	X
sg_result	1023
stallguard	*
fsactive	*
stst		
olb		*
ola		*
s2gb		*
s2ga		*
otpw		*
ot		*
Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!

Testing X connection... Error: All HIGH
</p>
</details>

<details>
<summary>M122 output (four drivers installed):</summary>
<p>
M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	795	795	795	795
MAX current	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual	31/31	31/31	31/31	31/31
PWM scale	255	255	255	255
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		0	0	0	0
[... 5011 lines cut ...]
 * Measures the filament width in real-time and adjusts
 * flow rate to compensate for any irregularities.
 *
 * Also allows the measured filament diameter to set the
 * extrusion rate, so the slicer only has to specify the
 * volume.
 *
 * Only a single extruder is supported at this time.
 *
 *  34 RAMPS_14    : Analog input 5 on the AUX2 connector
 *  81 PRINTRBOARD : Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 RAMBO       : Analog input 3
 *
 * Note: May require analog pins to be defined for other boards.
 */
//#define FILAMENT_WIDTH_SENSOR

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0    // Index of the extruder that has the filament sensor. :[0,1,2,3,4]
  #define MEASUREMENT_DELAY_CM        14    // (cm) The distance from the filament sensor to the melting chamber

  #define FILWIDTH_ERROR_MARGIN        1.0  // (mm) If a measurement differs too much from nominal width ignore it
  #define MAX_MEASUREMENT_DELAY       20    // (bytes) Buffer size for stored measurements (1 byte per cm). Must be larger than MEASUREMENT_DELAY_CM.

  #defi

Bruce17, 2020-04-28:
I found the issue: BigTreeTech TMC2130 v3.0 are **not** running SPI mode by default! They must be soldered from default STEP/DIR to SPI mode.

See instruction in repository https://github.com/bigtreetech/BIGTREETECH-TMC2130-V3.0 on how to do that.

With that adjustment, my TMC2130 is now finally recognized by marlin and M122 prints out something useful.

nickrodie, 2020-07-04:
> The TMC2130 requires that the 12v/24v power be active from power on, not later than the 5v.

OMG how was I supposed to diagnose that? Thank you so much for this solution.

Nonametruck, 2023-04-26:
Ok, I was having issues with my TMC2130's to run dual Z drivers. I found using the marlin 2.0 and taking the time in config.adv uncomitting use software spindle under the technology section and there is a spot where you tell how many z drivers you have put at 2. Also I found if I take all drivers out but x and y compile and do a m122 and check to see if those get a address then I shut down board put rest in and power up and m122 again and it seems to work fine now.  Skr v1.3