bigtreetech/BIGTREETECH-SKR-V1.3 #161: error message disappears

rvleeuwen92, 2019-12-08:
last week I equipped my tevo tarantula with the skr1.3 and touchscreen and it is almost working perfectly now. sometimes a print stops because of an error but the display shows the error message for quite a short time and then changes to the message Error:Printer halted. kill() called! is there any way to recall this error or to increase the time that is is showed on the display? Thanks in advance!

fxh1357, 2019-12-09:
The error may be send over the serial connection when attached to a host. So maybe connect to the printer using pronterface or so, while printing

rvleeuwen92, 2019-12-13:
@fxh1357 Thanks for your reaction! i tried to print with pronterface open and connected but unfortunately there is no message send over the serial connection when the error occurs.