bigtreetech/BIGTREETECH-SKR-V1.3 #162: MMU2

Sjoerrdd, 2019-12-09:
I'm trying to install the MMU2 (Marlin 2.0 supports it), I connected the connector to the TFT port, but can't make it work.
Which serialport is the TFT port? Has someone a solution?

Sjoerrdd, 2019-12-11:
I think I solved it, it was wired incorrectly. 
I'll try it tomorrow and let you know if it works :)

Jays3DAdventure, 2019-12-15:
How did you get on? Can you make your firmware available if did get it working please.

Thanks James.

Sjoerrdd, 2019-12-15:
@Jays3DAdventure https://github.com/Sjoerrdd/Jays3DAdventure
Make sure that you wired it correctly!

If you need help with the wiring, please drop a picture :)

Jays3DAdventure, 2019-12-15:
Link is 404 :(.

I'll check my wiring, thanks. Can you post a picture of your?

J.

Sjoerrdd, 2019-12-15:
[image]

closed 2019-12-17 by Sjoerrdd