bigtreetech/BIGTREETECH-SKR-V1.3 #163: Cant get Firmware to work

wuast94, 2019-12-14:
I have an skr 1.3 with TMC2208 V3.0. I only insert the jumpers on the Red UART pins. 

No matter what I do I only get errors on standard commands like M122:
Send: M122
Recv: echo:Unknown command: "M122"

I compiled the Firmware here and the one from marlin direct. 
I copied the 4 files in the ender3 example folder and the only things I edited are following: 

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 115200
  #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3

I followed the documentation here and from guides elsewhere. every time the same.

AnHardt, 2019-12-14:
How about configuring at least one stepper drive to be a TMC?

wuast94, 2019-12-14:
what exactly do you mean ?

AnHardt, 2019-12-14:
This is not a beginners forum. Please read the issue-template.

AnHardt, 2019-12-14:
Sorry. I assumed to be in the Marlin bugtecker.
M122 depends on the configured  existence of a TMC driver.
In the standard configurations they are not activated.
You have to activate at least one of them in the configurations to get a working M122.

wuast94, 2019-12-14:
ahh ok i will look for that. didn't see this in any guide so I wonder because there this commands worked too. or I missed something. I will give it a try later :)