bigtreetech/BIGTREETECH-SKR-V1.3 #166: skr 1.3 Bootloader

Marsman1970, 2019-12-18:
I murdered my poor Skr v1.3 board with a loose wire my fault Doh.  So i got a new Lpc1768 ic & other bits and fixed it. I was looking for a Bootloader i found this one 
https://github.com/Smoothieware/Smoothieware/tree/edge/bootloader
But it didn't enable Usb mode was it suppose to ? or did i miss something.  In the end i bought a new board and read the original Bootloader and put it in my Repaired board and it woks like new. Was there a place where i could have got the origonal file without getting a new spare board.
Cheers Michael

cjsoong, 2019-12-20:
Refer to this description
http://smoothieware.org/flashing-the-bootloader

Marsman1970, 2019-12-20:
I did all that but the bootloader they have there doesn't seem to be the same as the original on the board at least from when i tried it. Coping  it from a new Board was the same. That's why i though i ask because it was strange. To program i used a lpc isp mini V2 that worked great.

Interplain, 2020-01-06:
I have a Board that doesn't show a Mass storage drive or Com Port does this sound like the issue you had. The Red jumper is on USB and INT and it doesn't connect to the machine. Got a new board coming tomorrow. Because i don't know how to repair it. I have a Serial-USB but don't know what the hell its talking about. your a good one understanding that. Not the same as the Anet A8 board is it?

Marsman1970, 2020-01-07:
I got mine to connect to the pc but not mass storage. If you want to know about loading a boot loader it's a bit of a procedure, it took me a while to figure it out. There is 2 pins under the Z axis you need to short under the Z axis which says boot so the usb serial  works it's a bit tricky though. The  way i got it to work i used a screw driver to short the pins so you press the reset & short the boot pins then release reset & very quickly unshort boot pins.  It took me a while until Flash Magic could read the chip.

Interplain, 2020-01-07:
Right so you sorted lt with Windows.  So did you get the TF on the board to show up. After the flash?

Interplain, 2020-01-07:
And where did you get the Boot loader firmware from please.

Marsman1970, 2020-01-07:
Tf did work send me a email mher5454@yahoo.com.au 
and i can send you the original firmware i got of a new board as the the smoothiware didn't work so i got a new board & copied it to fix my old board.

Yuppi007, 2020-01-24:
Would you share the file. Because the original file is not in any source.

Interplain, 2020-01-24:
Yes Michael, I will do it when I get back home at work till 4 gmt but I
will do it as soon as possible.

Hope you are good.
Rich

Interplain, 2020-01-27:
Hi Michael, I don't seem to have the file. Don't think you sent it me. I
have looked in all emails you sent. Would you be able to save it again like
you did before by reading it from the working board.

Marsman1970, 2020-01-27:
I sent it if anyone else needs it email me. I haven't used github much but i should be able to upload it here  maybe it's time to learn lol.

Marsman1970, 2020-01-28:
Ok i uploaded it here try this. I did send it twice to your email but maybe it doesn't like hex files.
Cheers Michael
https://github.com/Marsman1970/SKR-V1.3-Bootloader

Yuppi007, 2020-01-28:
Thanks everyone.

Marsman1970, 2020-01-28:
So Yuppi007 it worked for you ok the boot loader file that is.
Michael

Pixel-Soup, 2020-01-31:
Hi Michael,
Thanks for uploading the bootloader.  Saved me the trouble of trying to find one.  It worked on my board, except for the USB port.  Win10 reports "device not recognized".

Marsman1970, 2020-01-31:
Usb should work as it,s a rip of a working board. 1 thing is if your Temperature sensors aren't correct or not connect it will disable the Usb.  If you still have issues i'll read a fresh copy of the hex file from my Board.
Cheers Michael

Pixel-Soup, 2020-01-31:
Thank you for the tips.  I found a microscopic sliver of solder shorting 2 of the mcu pins.  Now it works perfectly.

Interplain, 2020-01-31:
That's great you got it going, the workmanship is a little shoddy to say the least. But that is seen in the price of these boards. Thanks to Mars man's excellent help. Because the customer service is shite. I have a board waiting on a USB connection to flash this Hex file hopefully it will work. BIGTREE wanted me to send video evidence of a faulty board.. I sooner experiment with it to be honest. Pleased you got it going an it was just solder, I had a thought it would be that. Think my problem is bootloader.

Pixel-Soup, 2020-01-31:
In my case, I damaged the lpc1768, so I had to replaced it with a blank chip.  I can confirm that the USB drive and com port works with this boot loader after Marlin was loaded.  Everything is back to the way it was.

Marsman1970, 2020-02-01:
Good to hear it's working. Eventually i was looking at putting a Troubleshooting guide together for all the strange issues.

Interplain, 2020-02-11:
Hi Michael, hope you are ok I have got the FTDI USB serial cable how do I
connect it to the board can you circle the pins on a picture of the board
to show me where they go please. The cable has the plug on the end is it
the right one?

Thanks.

Marsman1970, 2020-02-11:
[SKR-V1.3-PIN.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4188950/SKR-V1.3-PIN.pdf)
It plugs into the TFT connector it's on the right of the Tf card socket. You need to connect RX & TX in reverse to each other  ie Rx to TX so when the FTDI USB serial cable transmits form the TX pin The skr board receives it on the RX pin.

Interplain, 2020-02-12:
Thanks Michael, I see where you mean now. It's the Aux 1 port an I will
have to separate the plug to connect to the correct pins on the board.  An
go through the boot process to get it found by the computer. An once found
by computer I can hopefully flash it.

dwillmore, 2020-03-19:
I just followed this procedure to replace a failed microcontroller on my SKR v1.3.  I would like to thank everyone in this thread for the wisdom that they have accumulated.  I would not have been able to do it without the information I found here.  Thank you, everyone.

N0mNus, 2020-05-15:
> [SKR-V1.3-PIN.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4188950/SKR-V1.3-PIN.pdf)
> It plugs into the TFT connector it's on the right of the Tf card socket. You need to connect RX [...]

which 2 pins ... do you have a photo =?

Ed-EE-Eng, 2020-05-20:
Say, I have this same problem with my SKR 1.4 Turbo.  
1) I initially was able to update the firmware with the SD card, but suddenly, the SD card reader was no longer recognized.  Doesn't show-up to print from nor do I see it when I connect a PC.
2) I can 'talk' to the the board with Octoprint and see status and print from OctoPrint.
3) Made some updates to Marlin and copied a new firmware.bin file to the SD care; Put SD care into SKR and nothing.  No Upload, firmware.bin remains on the care, no changes to FIRMWARE.CUR 

Appears either my SD card slot is defective or boot loader is gone.  

Does anyone know where I can get a bootloader file and detailed info on reflashing the boot loader onto a SKR 1.4 T?  TNX

codmpm, 2020-06-01:
I'm in the same boat as @Ed-EE-Eng. Exactly the same issue.
Thus I'm searching for the LPC176**9** bootloader to try to reflash it.

Does someone have the suitable bootloader for the SKR 1.4 Turbo?

Pixel-Soup, 2020-06-01:
@codmpm In your case, the 1769 and at least one of the drivers is blown.  A boot loader won't do any good.

codmpm, 2020-06-01:
Thank you! I've already changed the blown driver. Currently trying to revive the board via USB and without any drivers attached.
It will be a challenge, but I am able to change/rework the LPC1769. Till now I only did 48LQFP, but I think I can get it to work.

But after that I will still need a bootloader, didn't I?

Ed-EE-Eng, 2020-06-01:
So after playing with/inspecting my SKR board further, I am growing in the believe that I have a damaged SD socket.   It simply appears that the board can not read the SD card.  I've ordered some replacement SD sockets in the hopes that I can desolder and resolder a replacement.  

I also orderd and have just received an ST-LINK V2 programmer that supposedly can flash the 1769.  That will be another approach if I can't get the SD going.  Still looking for a boot loader...

Pixel-Soup, 2020-06-01:
@Ed-EE-Eng  My guess is that you uploaded firmware that did not properly support the SD pinout of the v1.4 board.  With the st-link, you should be able to load the original firmware onto the board without needing a boot loader. 
If you suspect the socket, you can try reinserting the card multiple times.  Or put a piece of tape on the back of the SD to push it harder against the contacts.

div1603, 2020-06-06:
Ed-EE-Eng
If you still need a bootloader file, I can provide it by e-mail.

div1603, 2020-06-06:
Ed-EE-Eng
You can download it from the link https://github.com/Smoothieware/Smoothieware/tree/edge/bootloader

11chuchu11, 2020-07-29:
hi i fucked my board's chip with a blown driver and i am waiting the replacement but i wanted to know if i could flash the chip with an arduino uno or any other and how

Ed-EE-Eng, 2020-08-03:
After screwing with my 1.4T + ST-LINK programmer for a couple of weeks, NO-JOY.  I don't think there are any "UNO" options.  I have set the board aside for now;  I have since bought a replacement 1.4T board - a brute force solution but it worked.  I will watch this thread for any other options...

dwillmore, 2020-08-03:
Are you talking about reprogramming the LPC chip?  You do that with a
simple USB<>serial converter (3.3V).  No special programmer needed.

codmpm, 2020-08-03:
I've successfully replaced the 1769 CPU and burnt the smoothieware bootloader with an segger j-link via SWD. Could'nt get the STM-V2 to work, though.

https://twitter.com/pregopm/status/1288855798223962112

Pixel-Soup, 2020-08-03:
@codmpm    It's interesting that the Smoothy bootloader works with skr v1.4.  It did not work on my skr v1.3
The LPC1769 is made by NXP.  The st-link software is made for STmicro's chips only.  That's likely why the st-link does not work.

codmpm, 2020-08-03:
@Pixel-Soup The SKR 1.3/SKR1.4 is using an LPC176**8**, thus the smoothieware (LPC176**9**) bootloader will not work--- the 1.4 Turbo uses the LPC1769, same as the Smoothieboard (IIRC).

Marsman1970, 2020-08-03:
The SKR 1.3 bootloader i uploaded is for a LPC1768 straight from a 1.3 board. The smoothieware one that's uploaded here didn't work for me either so i got a new board to copy it from & fix my board.
https://github.com/Marsman1970/SKR-V1.3-Bootloader

ShadowOfTheDamn, 2020-11-09:
Add me to those who flashed the firmware on skr1.4 turbo with Segger J-Link with SWD.

GadgetAngel, 2021-01-18:
I am trying to create a repository for all the BIGTREETECH boards and their bootloaders for each board, here: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader