bigtreetech/BIGTREETECH-SKR-V1.3 #173: SKR 1.3 TMC2209 noise

elmolinger, 2019-12-23:
Hi there,

I have installed a SKR1.3 with TMC2209 stepper drivers. The steppers make a lot of noise when printing. I tried to set the current without success. If Stealtchop is deactivated, no difference can be seen either.

It sounds like the printer is accelerating violently and stuttering.
In a circular motion, the steppers make a noise that I know from the old a4498 drivers, but I don't expect that from the TMC2209. Before that, a MKS 1.4 8-bit board with TMC 2208 drivers was installed, which was a lot quieter with the same settings.

See video
https://www.youtube.com/watch?v=bcKDEgzNxB4

I am using a current Marlin 2.0bugfix from today.
The TMC library used is 0.6.1

[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3995743/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/3995744/Configuration_adv.h.txt)

Any help would be appreciate

AndKe, 2020-01-24:
did you set the jumpers/configure it correctly ? https://www.youtube.com/watch?v=H5ea2iyOscg

freighttt, 2020-06-19:
@elmolinger did you ever get this sorted out? i am having the same issue

elmolinger, 2020-06-19:
@freighttt no i didnt. But im very sure its about the 0.9 degree stepper. I disassembled the Printer just 3days ago and build a new one. Now with a SKR1.4 (LPC1768) in the new Printer and 1.8degree Stepper Motors. Printing without noise and any Issues.

Im pretty sure it was about the 0.9Degree stepper motors and the selfmade Extruder Carriage on the X Axis. Wich causes a mechanical issue.

Wich Stepper Motors do you have in Use?

freighttt, 2020-06-20:
I’m using ldo 42sth47-1684MAC(I think I got that right), basically 0.9* steppers running at 1050ma on a corexy. my issue actually isn’t like yours, I didn’t watch the video you posted until just now. My issue is stealthchop making extremely loud grinding motor noise above ~35mm/s, but running fine up past 200mm/s in spreadcycle. Stealthchop is just as loud as spreadcycle, like in your case

elmolinger, 2020-06-20:
My driver was also running stealtchop in this video ;-) 

It seems there is an issue with tmc2209 and 0.9 degrees stepper motors. My stepper motors was e3d steppers and also 0.9 degree.

Can you try to use a 1.8degree stepper on one axis and try to print a cube in vase mode so you can figure this out. That was my planned next step with the old printer, but due difficulty to disassemble the frame I never got it done.

h3po, 2020-06-30:
related marlin ticket: https://github.com/MarlinFirmware/Marlin/issues/18286