bigtreetech/BIGTREETECH-SKR-V1.3 #175: Problems with SKR1.4 an Dual-Endstops

NikNolte, 2019-12-26:
**There are Problems with the Firmware SKR1.4 and Dual-Endstops:

My Endstop settings in Configuration.h**

// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_ZMAX_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  #define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Enable pulldown for all endstops to prevent a floating state
//#define ENDSTOPPULLDOWNS
#if DISABLED(ENDSTOPPULLDOWNS)
  // Disable ENDSTOPPULLDOWNS to set pulldowns individually
  //#define ENDSTOPPULLDOWN_XMAX
  //#define ENDSTOPPULLDOWN_YMAX
  //#define ENDSTOPPULLDOWN_ZMAX
  //#define ENDSTOPPULLDOWN_XMIN
  //#define ENDSTOPPULLDOWN_YMIN
  //#define ENDSTOPPULLDOWN_ZMIN
  //#define ENDSTOPPULLDOWN_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
//#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
//#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.

**My Dual-Z-Endstop settings:** in Configuration_adv.h**

#define Z_DUAL_STEPPER_DRIVERS
#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
  #define Z_DUAL_ENDSTOPS
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _ZMAX_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif

When I compile this, I get always the same error messages

**#error "Z2_USE_ENDSTOP has been assigned to a nonexistent endstop!"
      ^~~~~
In file included from d:\marlin-dev-2.1.x\marlin\src\inc\marlinconfig.h:38,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
d:\marlin-dev-2.1.x\marlin\src\inc\SanityCheck.h:1766:6: error: #error "Z2_USE_ENDSTOP has been assigned to a nonexistent endstop!"
     #error "Z2_USE_ENDSTOP has been assigned to a nonexistent endstop!"**

I have this Proble not with the SKR1.3..
Can you help me ??

Neme77, 2020-01-10:
The problem is in the pin definition file: **pins_BTT_SKR_V1_4.h**,
the definitions of the limit switches are missing, more precisely they are incomplete.
To fix them you need to add the complete definition like there:

// Limit Switches
//
#define X_MIN_PIN          P1_29
#define X_MAX_PIN          P1_26
#define Y_MIN_PIN          P1_28
#define Y_MAX_PIN          P1_25
#define Z_MIN_PIN          P1_27
#define Z_MAX_PIN          P1_00

NikNolte, 2020-01-11:
OK... and whats about the pins for the Stepper E1 ??
I need two Stepper for the Z-Axis ..

lijinlin19970813, 2020-01-14:
> 问题出在引脚定义文件：**pins_BTT_SKR_V1_4.h中**，
> 限位开关的定义丢失了，更确切地说，它们是不完整的。
> 要修复它们，您需要添加完整的定义，例如：
> 
> //限位开关
> //
> #define X_MIN_PIN P1_29
> #define X_MAX_PIN P1_26
> #define Y_MIN_PIN P1_28
> #define Y_MAX_PIN P1_25
> #define Z_MIN_PIN P1_27
> #define Z_MAX_PIN P1_00

Because the motherboard's XYZ_MAX is E0DET, E1DET, PWRDET, you will find that the corresponding pins in the pin file are changed to other function pins. If you want to use it, you can change it yourself.

OleUrgast, 2020-01-29:
Of course you shoud not only add #define Z_MAX_PIN P1_00 but also comment out
//#define PS_ON_PIN        P1_00
and
#define POWER_LOSS_PIN   P1_00

If you use dual z with two endstops, you can only connect one extruder anyway, as you use e1 stepper for second z motor.
So I recommend to use the (in this case) useless filament runout pin for a second extruder (E1DET) for z_max:
- add "#define Z_MAX_PIN P1_25
- comment out "//#define FIL_RUNOUT2_PIN    P1_25"
and connect second z-endstop to E1DET

So you do not loose power detection feature...

jimprince, 2020-03-18:
It seems there isn't a dedicated _xmin_/_xmax_ on the skr1.4, they're both just the "X endstop." I had to add a definition for xmin and max as stated by @Neme77 above, and then ignore a TON of compile time warnings to get dual endstops working.

IIIxpHunterIII, 2020-10-09:
For me it rips the Z1 Stepper to leadscrew connector apart with that.

#define X_MIN_PIN P1_29
#define X_MAX_PIN P1_26
#define Y_MIN_PIN P1_28
#define Y_MAX_PIN P1_25
#define Z_MIN_PIN P1_27
#define Z_MAX_PIN P1_00