bigtreetech/BIGTREETECH-SKR-V1.3 #177: No firmware for SKR 1.4 turbo!

Mosher23, 2019-12-29:
so i have checked the repository and found the the only firmware available is for skr 1.4 with LPC1768, but no fw for skr 1.4 turbo with LPC1769! simply changing mcu in platfromio.io file will, not work as marlin has no pins defined for skr 1.4 with lpc1769.
What should i do?

manuelchaves, 2019-12-29:
Hi.
In fact SKR like any other application that uses the LPC1769 / 68/67/66/65/64/63 family as a processor, the drivers are the same, which differentiates in the case of turbo is only the processing speed ( clock). And if it were any other of the same processor family the serial differences in the encapsulation and in some like 1765 they do not have etehrnet.
That is, when selecting the FW LPC1768 it automatically opens the whole LPC1769 / 68/67/66/65/64/63 family.
Hope this helps.
See this link to the LPC1769 / 68/67/66/65/64/63 family datasheet https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf

delwinbest, 2020-02-02:
@Mosher23 here's an example of a working Marlin build for the SKR 1.4 Turbo:
https://github.com/delwinbest/Marlin_BIBO2_touch_dual/tree/btt_skr_v1.4_turbo

It's a branch of the Marlin 2.0 mainline, I use it on my BIBO2 Touch Printer

77katz77, 2020-02-09:
Has anyone been able to get a driver working for the 1.4 Turbo and been able to communicate with it not just accessing the sd card? I've tried using the lpc176x driver included in Marlin. It seems however that only the 1768 is defined in the driver not 1769.

Mosher23, 2020-02-09:
@77katz77 my skr 1.4 turbo working flawlessly with my tmc5160s