bigtreetech/BIGTREETECH-SKR-V1.3 #179: ESP-01 setup

K2AAE, 2019-12-30:
It is nice there is a socket for it but how do you get it to work?

OleUrgast, 2020-01-29:
https://github.com/bigtreetech/BTT-WRITER-V1.0

rainerschulte, 2020-02-23:
I think it's asimilar problem to me. A dedicated Socket is available on SKR1.4 or in my case on the TFT35 (BTT). My Printer (SKR1.3 and TFT35) is up and running. I tried to expand my TFT with a ESP01-S, which is flashed and setup with ESP3D. The ESP works fine, as long it is plugged into a USB-Programmer (not in Programm-Mode). I can access the ESP via IP from any computer in my Network. 
But when I plug the ESP to my TFT, the ESP will not be found in my Router and cannot get reached/accessed by IP. The TFT itself doesn't seem to recognize anything, because "M117" doesn't come up too, when powering the Printer on. 

My assumption is, that there is a little hook left to loose in Marlin. But I have no clue what...

OleUrgast, 2020-02-23:
> My assumption is, that there is a little hook left to loose in Marlin. But I have no clue what...
If Wifi module is attached to TFT, Marlin has nothing to do with it. The display is like a seperate computer. It simply send gcode to the mainboard. The diplaycontroller is responsible to controll the SD, USB and wifi module. So you should look in code, description and discussion of the BTT TFT 35. It has nothing to do with SKR 1.3/1.4 and Marlin.

rudranshsinghmahra, 2024-10-07:
> I think it's asimilar problem to me. A dedicated Socket is available on SKR1.4 or in my case on the TFT35 (BTT). My Printer (SKR1.3 and TFT35) is up and running. I tried to expand my TFT with a [...]

Did you find the solution to this. I am facing the exactly same issue. The difference is that I have plugged my ESP01 Module direct to the my SKR 1.4 dedicated WIFI Socket provided.